mavlink/qgroundcontrol #10578: Starting QGroundControl AppImage gives symbol lookup error

SemvdH, 2023-02-15:
## Expected Behavior
I'm trying to launch the QGroundControl AppImage from the terminal (using `./QGroundControl.AppImage`).

## Current Behavior
When I enter the command `./QGroundControl.AppImage` I get the following error message:
`/tmp/.mount_QGroun6DvwYI/QGroundControl: symbol lookup error: /tmp/.mount_QGroun6DvwYI/QGroundControl: undefined symbol: _ZNSt13runtime_errorC1ERKS_, version Qt_5`

## Steps to Reproduce:
1. Follow the [Ubuntu install instructions](https://docs.qgroundcontrol.com/master/en/getting_started/download_and_install.html#ubuntu) from the QGroundControl website
2. log out and log back in
3. Download the [v4.2.4 AppImage](https://github.com/mavlink/qgroundcontrol/releases/tag/v4.2.4)
4. make it executable
5. execute the command `./QGroundControl.AppImage`

## System Information
- Operating System: Ubuntu 22.04.1 LTS x86_64
- QGC Version: 4.2.4
- QGC build: stable
- Flight Controller: PixHawk 4 Mini
- Autopilot (with version): PX4 1.13.1

## Detailed Description
I have installed all the necessary libraries as shown on the install page of QGroundControl. When I try to launch the AppImage from [v4.0.0](https://github.com/mavlink/qgroundcontrol/releases/tag/v4.0.0) I do not get the error.

SemvdH, 2023-02-15:
I just tried with [v4.1.5](https://github.com/mavlink/qgroundcontrol/releases/tag/v4.1.5), and on that version I don't get the error.

SemvdH, 2023-02-15:
Up to [v4.1.7](https://github.com/mavlink/qgroundcontrol/releases/tag/v4.1.7) works, after that, I get the error

booo, 2023-02-15:
Could you check a unstable/latest build as well?

SemvdH, 2023-03-06:
Sorry for the late response, I got caught up in other stuff. I tried the daily build from [here](https://docs.qgroundcontrol.com/master/en/releases/daily_builds.html) but I get the same error: `/tmp/.mount_QGrounamQWEG/QGroundControl: symbol lookup error: /tmp/.mount_QGrounamQWEG/QGroundControl: undefined symbol: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEED1Ev, version Qt_5`

HTRamsey, 2024-05-31:
AppImage creation is significantly different now. Please reopen if you still have any issues.

closed 2024-05-31 by HTRamsey