mavlink/qgroundcontrol #10580: Survey at an angle:  Add option to fly a survey at an angle.

AndKe, 2023-02-17:
Some sensors do not look forward but to the side.  This feature is useful for side-scanning along a grid.

Basically; take normal Survey item:
[image]

and add the heading fields from WAIT_FOR_YAW mission item. (which is MAV_CMD_CONDITION_YAW)
[image]

Mission generation logic:
After each waypoint - a "MAV_CMD_CONDITION_YAW" mission item needs to be added.
https://mavlink.io/en/messages/common.html#MAV_CMD_CONDITION_YAW

We assume one heading during the whole mission, so the user-editable fields should be:
1: Angle(0-359°)
2: Angular Speed default 10° - may be useful for different vehicles
these parameters should be static:
3: Direction (0 = always the shortest way - no need to change, the whole mission will face the same direction)
4: Absolute angle (just keep a given heading.

AndKe, 2023-05-08:
@DonLakeFlyer  Please let me know if any other feature of the survey generates additional mission items, and has some GUI items, preferably added in one commit.   
If I could use such a previous commit as an example, then maybe I could add this feature myself.

DonLakeFlyer, 2023-05-21:
Yes and no. For interior waypoints the output is a single item, but for say survey edge waypoints they can output multiple items. If you look at TransectStyleComplexItem::_buildAndAppendMissionItems you'll see how transects are built. You could theoretically modify this to add multiple items to interior waypoints as well. Not simple change, but not impossible either.

closed 2026-10-04 by github-actions[bot]