mavlink/qgroundcontrol #10581: Desktop and Android version don't match view

TheLastImperial, 2023-02-17:
The following template is for Issue/Bug reporting only. https://docs.qgroundcontrol.com/en/Support/Support.html#github-issues

For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources

If you are using an old stable release please upgrade first and verify your issue before reporting.
----

## Expected Behavior
The same view to desktop and android version.

## Current Behavior

The android view look like don't

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. I follow the [guide](https://dev.qgroundcontrol.com/master/en/getting_started/index.html)
2. Compile to desktop
3. Compile to android

## System Information
When posting bug reports, include the following information
- Operating System: Ubuntu 20.04
- QGC Version: Stable_V4.2
- QGC build: Stable_V4.2
- Flight Controller: Pixhawk 4
- Autopilot (with version): [e.g., PX4 1.8.1, Ardupilot Copter 3.6.1]

## Detailed Description

Desktop version look like this.

[image]

The android version:

[image]

When I go to the video source don't show options.

[image]

## Log Files and Screenshots

[appOutput.txt](https://github.com/mavlink/qgroundcontrol/files/10771432/appOutput.txt)

[compileOutput.txt](https://github.com/mavlink/qgroundcontrol/files/10771433/compileOutput.txt)

The only strange that I can watch is in the appOutput.txt.

```txt
W System.err: java.lang.ClassNotFoundException: Didn't find class "org.qtproject.qt5.android.bearer.QtNetworkReceiver" on path: DexPathList[[],nativeLibraryDirectories=[/system/lib64]]
```

But I realy don't know if this is the problem.

TheLastImperial, 2023-02-18:
Well. I think that I find the problem. I am using the SIYI MK15 control and the preinstalled  QGC work fine but when I build a custom version look like  the above images and when I install the custom QGC in my phone I have this:

Main view:
[image]

Settings view:
[image]

Conclusion I need the custom version from SIYI to build a custom version to my MK15 control.

Davidsastresas, 2023-02-19:
Please keep in mind in order to build android with video support you need to have gstreamer on your project path, there is some info here:

https://github.com/mavlink/qgroundcontrol/blob/master/src/VideoReceiver/README.md

It could be for sure that custom SIYI QGC has something else besides this... But check the above as it might help. 

I am closing this one as I think it is out of the scope of the project to support closed source forks of QGC.

closed 2023-02-19 by Davidsastresas