mavlink/qgroundcontrol #10592: Offline Maps: no elevation for northern Norway/Svalbard.

AndKe, 2023-02-21:
## Expected Behavior
I would expect elevation data just as for the rest of the world.

## Current Behavior
They won't be downloaded.: 
[image]

## Steps to Reproduce:
1. in planning, or maps caching - observe missing data.

## System Information
- QGC Version: current master

## Detailed Description
It's been like this for a long time

## Log Files and Screenshots
- [QGC Console Logs](https://docs.qgroundcontrol.com/en/Support/Support.html#reporting-bugs)
- Autopilot logs when available (post a link).
- Screenshots of QGC to help identify the current issue/bug behavior.

booo, 2023-02-25:
As far as I know there are areas of the world where the model does not provide elevation data. Could you please double check on that. Thanks a lot.

AndKe, 2023-02-26:
@booo yes, I've tried random spots in Scandinavia, Europe (I think Germany or Poland) and USA and assumed this is a wide issue.  - Is there a debug-log flag for elevation-fetching?

booo, 2023-02-26:
The elevation data download is broken for other spots of Europe too?

https://terrain.ardupilot.org/ here you can download terrain data manual for ardupilot.

Sorry, I'm not familiar with this part of the code. If there is no logging maybe you can add some in a PR?

closed 2024-04-06 by DonLakeFlyer