mavlink/qgroundcontrol #10593: Offline maps: cosmetic issues.

AndKe, 2023-02-21:
Tiles count does not fit:

[image]

Apart from "default map" - only the last download gets a green dot - all other are red.  - the "green dot" - whitch I guess means "completed/fetched" is ...not stored.
[image]

There is tons of screen area, no need to use such a narrow column.
[image]

Davidsastresas, 2023-02-22:
This was addressed here: #10598 

It should fix the text overlapping. For the buttons going green or not, they will go green when the whole tileset is downloaded. From your screenshot it seems the tilesets labeled in red have not been completely downloaded, thus they are red.

On my screenshots on the PR you can see it worked nicely for me. Maybe you got bad connection or something and they were not able to complete.

Please report back and close the issue if the PR fixes it for you. Thanks for reporting!

AndKe, 2023-02-22:
@Davidsastresas please let me show you:

[Screencast from 2023-02-22 23-08-20.webm](https://user-images.githubusercontent.com/1468842/220771163-bb2b0781-71c3-47d3-9972-be14a2f73a43.webm)

In other words: after download - they are green and fine, successfully completed. Then I restart QGC, and all are red - but eager to become green is "resumed" with 0 tiles.  Hence my idea about the "dot-status" not being stored.

Davidsastresas, 2023-02-22:
Sorry, I didn't understood correctly. Just tested, it is working fine for me. It is funny because a couple of months back a partner of mine reported weird stuff happening on QGC and ubuntu 22. We took a look real quick, it wasn't making sense the time we were putting into it so he just rolled back to 20.04 in order to keep working. I just tested on my machine, ubuntu 20.04 and it is working as expected:

https://drive.google.com/file/d/1iwMxc4eBWqv_Oi_mb3pRkKxDXps-4soL/view?usp=sharing

So maybe there is something in ubuntu 22 that is conflicting with QGC some way.

AndKe, 2023-02-22:
@Davidsastresas  I do believe you .. but we both know how extremely odd this sounds..  of millions of CPU cycles, some get to result in a different outcome?  :)   - this is bizarre ..and scary  :o

AndKe, 2023-02-22:
Additional information: latest QGC .AppImage behaves the same way.
I am aware of some SSL issues on Ubuntu22.10 that only manifest themself on APMplanner2 and QGC, and in case of those, Appimage works fine.  (of course, I don't see how this simple thing could be SSL related) - the tiles are after all downloaded just fine.

Davidsastresas, 2023-02-23:
It is extremely odd indeed. If more people get to confirm this, maybe we should close this issue for the cosmetic changes, and open a new issue with all the issues together related to ubuntu 22?

My machine is on ubuntu 20.04 now. I want to update to 22, but I don't think I will spare the time to do so on the short future, so I don't think I will be able to take a look at this on the following weeks.

AndKe, 2023-02-23:
@Davidsastresas Where are my tiles /tile-db stored? - I should try to delete it in case this is a corrupted setup (I did export/import by tiles-db and synced across computers before - maybe I synced the error)
I can try to delete/move it for a test.
I would expect the tiles to be stored in one of those dirs:
"~/Documents/QGroundControl Daily"
"~/Documents/QGroundControl"
...but to my surprise ... they are not there.

Davidsastresas, 2023-02-24:
In my Linux machine they are located here:

home/"user"/.cache/QGCMapCache300

Let me know if you advance on this. Thanks!

AndKe, 2023-02-25:
@Davidsastresas 
I renamed ~/.cache/QGroundControl.org and ~/.cache/QGCMapCache300   - and the status dot behaves as expected.
if this is of any interest to fix this non-critical bug - I could pack up and share those directories. 

BTW: fetching elevation fails consistently every time :)  - is "airmap services" required to fetch elevation data ?

booo, 2023-03-21:
I don't think "airmap services" are required to fetch elevation data.

I guess we don't catch an error at the right spot. Do you have time to dig in? Deleting the cache is fine but a proper solution would be robust code.

AndKe, 2023-04-10:
I tried to look into it, but my Qt skills are insufficient for proper debugging.

HTRamsey, 2024-05-31:
This part of the UI has largely been rewritten

closed 2024-05-31 by HTRamsey