mavlink/qgroundcontrol #10595: Vehicle.cc : / MAVLink  writing 50 bytes into a region of size 24

AndKe, 2023-02-22:
Compiler warning on current master. 
This can cause issues in case of a odd package,
```
In function ‘strncpy’,
    inlined from ‘_handleStatusText’ at /home/andre/prog/qgroundcontrol/src/Vehicle/Vehicle.cc:950:12,
    inlined from ‘_mavlinkMessageReceived’ at /home/andre/prog/qgroundcontrol/src/Vehicle/Vehicle.cc:751:26:
/usr/include/x86_64-linux-gnu/bits/string_fortified.h:95:34: warning: ‘__builtin_strncpy’ writing 50 bytes into a region of size 24 overflows the destination [-Wstringop-overflow=]
   95 |   return __builtin___strncpy_chk (__dest, __src, __len,
      |                                  ^
/home/andre/Qt/5.15.2/gcc_64/include/QtCore/qarraydata.h: In member function ‘_mavlinkMessageReceived’:
/home/andre/Qt/5.15.2/gcc_64/include/QtCore/qarraydata.h:129:8: note: destination object ‘<anonymous>’ of size 24
  129 | struct QTypedArrayData
      |        ^

```

Davidsastresas, 2023-02-22:
I just took a look into this, I can not reproduce the compiler warning. Checking the code, we have this is the problematic piece of code:

    QByteArray  b;
    QString     messageText;

    mavlink_statustext_t statustext;
    mavlink_msg_statustext_decode(&message, &statustext);

    uint8_t compId = message.compid;

    b.resize(MAVLINK_MSG_STATUSTEXT_FIELD_TEXT_LEN+1);
    strncpy(b.data(), statustext.text, MAVLINK_MSG_STATUSTEXT_FIELD_TEXT_LEN);

where b.data() is the destination, which is of size MAVLINK_MSG_STATUSTEXT_FIELD_TEXT_LEN+1, one more than the source, statustext.text, which is a char[50].

What is your setup for building this? I just rebased to master and built and I didn't receive this warning. Please tell us more about it.

Thanks.

Davidsastresas, 2023-02-22:
Addressed here #10597. 

I wasn't having this error even before that PR. What is your development environment? Please report back and close this issue if this fix the warning. Thanks for reporting!

closed 2024-05-31 by HTRamsey