mavlink/qgroundcontrol #10601: Add support for better battery information, similar to dji battery bar thingy

DonLakeFlyer, 2023-02-25:
Information from MaEtUgR on discord:
sure, I have to write a revised battery documentation but in short for the ground station interface the MAVLink message https://mavlink.io/en/messages/common.html#BATTERY_STATUS contains: 

* battery_remaining autopilot's estimated percentage (if it's not a smart battery) already shown in QGC 
* time_remaining autopilot's estimate of remaining flight/run time considering configured battery capacity, filtered current consumption and the battery estimate (only available if e.g. BAT1_CAPACITY is configured, be careful misconfiguration leads to an unusable battery estimate) 
* charge_state autopilot's current judgement of the battery state which triggers low battery actions that are configured to a fixed percentage e.g. should probably be shown as obvious green, orange, red UI to the user https://mavlink.io/en/messages/common.html#TIME_ESTIMATE_TO_TARGET contains safe_return which is the autopilot's current estimate for how long it takes to execute a return to home or closest rally point if configured with some margin. When this time goes below BATTERY_STATUS.time_remaining an RTL is triggered by default. This allows to show a bar similar to the DJI app where you see live how much time is left until it happens.

DonLakeFlyer, 2023-02-26:
Also @MaEtUgR: In the mavlink docs this is still marked Work In Progress. Is this only available in master and not out in a stable yet?

MaEtUgR, 2023-02-27:
> In the mavlink docs this is still marked Work In Progress. Is this only available in master and not out in a stable yet?

@DonLakeFlyer The big part is in stable 1.13 (or even before):
- The low battery conditions are published for quite some time also in 1.13: https://github.com/PX4/PX4-Autopilot/blob/release/1.13/src/modules/mavlink/streams/BATTERY_STATUS.hpp#L83-L115
- Remaining battery time as well: https://github.com/PX4/PX4-Autopilot/blob/release/1.13/src/modules/mavlink/streams/BATTERY_STATUS.hpp#L80-L81

The only thing that will newly get available in 1.14 is the information on how long it takes for the vehicle to return `TIME_ESTIMATE_TO_TARGET.safe_return` : https://github.com/PX4/PX4-Autopilot/blob/release/1.14/src/modules/mavlink/streams/TIME_ESTIMATE_TO_TARGET.hpp#L67
Before this message is not streamed at all.

MaEtUgR, 2023-02-27:
Also note that if you want to test the remaining flight time in simulation it's not really easily accessible because it simulates no current and hence purely voltage-based estimation. I can make a pr where you can just enable it in the simulated estimation.

DonLakeFlyer, 2023-03-01:
> I can make a pr where you can just enable it in the simulated estimation.

Yeah. That would be helpful.

The problem is that as far as I can tell on TIME_ESIMATE_TO_TARGET will work for a battery bar thingy. Otherwise you get into the whole multi-battery mess and what do you do if you have multiple batteries reporting status?

DonLakeFlyer, 2026-03-26:
Cant be done with info currently available as far as I know