mavlink/qgroundcontrol #10606: Improve map source definition and add support for other projections.

AndKe, 2023-02-27:
MAVProxy has this super-clean way of adding maps - one line, done. 
Defining a new map in QGC takes 20 lines across three files.

Please add support for other ESPG (projections) too.

- What problem does it solve?
This allows for adding some very detailed maps. - many are other/local ESPG types.
This would also clean up the way a map is added.

bpmooch, 2023-03-01:
@AndKe Do you have a link to the relevant file in MAVProxy?

AndKe, 2023-03-01:
@bpmooch  - please see https://github.com/ArduPilot/MAVProxy/blob/master/MAVProxy/modules/mavproxy_map/mp_tile.py#L69
at line 69, you see all it takes to add a map.  - (but only one one projection is supported)