mavlink/qgroundcontrol #10608: Need help moving QGCInstrumentWidget to a new window in QGroundControl

AnaabTeam, 2023-03-01:
I'm trying to move the QGCInstrumentWidget from the main window to a new window in QGroundControl, but I'm not sure how to do it. I've tried a few different approaches, but nothing seems to be working.

Could someone please help me with this issue? I would greatly appreciate any guidance or suggestions on how to achieve this.

[image]

zdanek, 2023-03-01:
Hi.
For better understanding what is the purpose?
Do you want to make a PR with it so it will get into QGC official code or you want to do it for yourself?

AnaabTeam, 2023-03-01:
Hi,

Thanks for getting back to me. My goal is to move the QGCInstrumentWidget to a new window for my own use, as it would make it easier for me to work with. I don't currently have any plans to make a pull request with the changes, but if the changes turn out to be useful and beneficial to others, I would be open to it.

I appreciate any help or guidance you can offer on this issue. Thank you!

Davidsastresas, 2023-03-01:
If I understood correctly, you want it to be "detachable"? So you can move it to a different screen? If so, QGC does that for the mavlink inspector view, and also fot the video widget. If you pull that string I am sure you will be able to figure it out.

AnaabTeam, 2023-03-02:
[image]

booo, 2023-03-21:
Closing this because it not a bug and not a feature request and the questions is answered with a few hints in the right direction. For further question you can also use the discord chat.