mavlink/qgroundcontrol #10609: Serious bug destroys planned mission by making it un-editable further by QGC

AndKe, 2023-03-01:
## Expected Behavior
I'd expect the QGC not suddenly decide that the whole mission is in one reference frame.

## Current Behavior
Saving .plan (only supported format) can result in a mission that is unable to be edited when it comes to altitude frame.

## Steps to Reproduce:
1: Download and unpack (because github does not allow *.plan files)
[Badmission.zip](https://github.com/mavlink/qgroundcontrol/files/10864390/Badmission.zip)

2.load in in QGC 
3.observe that at acts as if the whole mission is in one kind of altitude(relative)
[image]

4....check the file: there are many terrain-frame waypoints.
5 act as if you need to add one mode, terrain-frame waypoint - impossible... you are unable to work on it anymore (in the correct altitude frame). 

## System Information
When posting bug reports, include the following information
- Operating System: Ubuntu
- QGC build: master , last release ,  .Appimage

## Detailed Description
this mission is manually drawn due to complex terrain and winds.
Due to its sensitive nature, it is moved to nowhere. 

Once a mission snags like this - it causes unnecessary problems.  Please fix the logic - it should at least allow changing a mission to "mixed mode" anytime.
having the logic fail like this, then "lock up" in "terrain mode" is just bad.  

## Workaround
Now that QGC sabotaged this fine-tuned mission, the only solution was to upload it to the copter, then use another GCS to download it - and continue working on it.

AndKe, 2023-03-01:
@Davidsastresas  Thank you.
To me, it seems like the terrain frame, (which is mostly just another uint_8t integer value) is like kryptonite to QGC. 
There are at least three more issues that are tightly related to "terrain frame" that I suspect can be related:
https://github.com/mavlink/qgroundcontrol/issues/10596
https://github.com/mavlink/qgroundcontrol/issues/10557
https://github.com/mavlink/qgroundcontrol/issues/10566

with some luck, figuring out one of them may solve more than one issue.
 
Also: I don't understand the .plan format, it contains too much unnecessary/redundant data:
            {
                "AMSLAltAboveTerrain": null,      << What is the use of this field - it's certainly irrelevant to most commands.
                "Altitude": 16.5,                            <<redundant , as altitude is anyway in param7 
                "AltitudeMode": 4,                        << how on earth did altitude-mode/coordinate/frame 10 become 4?
                "autoContinue": true,                   << ? ...what is this expected to do? AP will always continue to the next cmd.
                "command": 16,                   
                "doJumpId": 55,
                "frame": 10,
                "params": [
                    0,
                    0,
                    0,
                    0,
                    80.47098059999999,
                    15.0016223,
                    16.5
                ],
            {
                "AMSLAltAboveTerrain": null,
                "Altitude": 16.5,
                "AltitudeMode": 4,
                "autoContinue": true,
                "command": 16,
                "doJumpId": 55,
                "frame": 10,
                "params": [
                    0,
                    0,
                    0,
                    0,
                    80.47098059999999,
                    15.0016223,
                    16.5
                ],
                "type": "SimpleItem"
            },
I am not expecting questions to the above, just pointing out that there are many pitfalls around having this much odd and 
redundant information, and "translating" 10 to 4 in between.

I would suggest an extended QGC WPL 110(like) format, with commented data that supporting systems could read, but where the mission items still were stored in an unambiguous way.

Davidsastresas, 2023-03-02:
I think the part generating .plan files has been there for a long time, maybe it is a good moment to take a look at them and update them. I am not super familiarized with that part right now, so I can not comment much more.

Regarding terrain frame, indeed until kind of recently, terrain frame support was not complete in QGC, so there has been work lately ( I think last year ) about this, I recall seeing the changes at some point. Maybe something was left off.

I will take a look at it ASAP. It is the third item on my list of priorities, hopefully on the next few weeks I will be able to go over it. Thanks!