mavlink/qgroundcontrol #10613: Container build error - PermissionError: [Errno 13] Permission denied: '/usr/lib/python3.8

amokduke, 2023-03-08:
----

## Expected Behavior
Following the steps listed at https://dev.qgroundcontrol.com/master/en/getting_started/container.html to build QGC.
Expect that the build container got built successfully. 

## Current Behavior
There is an error while building the container...
PermissionError: [Errno 13] Permission denied: '/usr/lib/python3.8/__pycache__/__future__.cpython-38.pyc.139795054894752'

## Steps to Reproduce:
1. Install docker using steps here: https://docs.docker.com/engine/install/ubuntu/
2. Using commit "Update PX4 Firmware metadata Fri Dec 30 06:49:42 UTC 2022"
3. docker build --file ./deploy/docker/Dockerfile-build-linux -t qgc-linux-docker .

## System Information
- Operating System: [Ubuntu 22.04.1 LTS]
- QGC sourcecode: 
[image]

## Detailed Description
Have reinstalled docker and remove all images but to no avail.

## Log Files and Screenshots
[image]

booo, 2023-03-21:
I'm not sure how well maintained this part of the codebase actually is.

I think you have problem with the packages the docker image tries to install. Please try to find the packages I guess you need another python-minimal package.

davidmoyalopez, 2023-03-27:
Hello,
I have the same error.

Error:
```
Selecting previously unselected package libpython3.7-minimal:amd64.
Preparing to unpack .../4-libpython3.7-minimal_3.7.3-2+deb10u4_amd64.deb ...
Unpacking libpython3.7-minimal:amd64 (3.7.3-2+deb10u4) ...
Preparing to unpack .../5-libexpat1_2.2.6-2+deb10u6_amd64.deb ...
Unpacking libexpat1:amd64 (2.2.6-2+deb10u6) over (2.2.6-2+deb10u4) ...
Selecting previously unselected package python3.7-minimal.
Preparing to unpack .../6-python3.7-minimal_3.7.3-2+deb10u4_amd64.deb ...
Unpacking python3.7-minimal (3.7.3-2+deb10u4) ...
Setting up libpython3.7-minimal:amd64 (3.7.3-2+deb10u4) ...
Setting up libexpat1:amd64 (2.2.6-2+deb10u6) ...
Setting up python3.7-minimal (3.7.3-2+deb10u4) ...
Traceback (most recent call last):
  File "/usr/lib/python3.7/py_compile.py", line 212, in <module>
    sys.exit(main())
  File "/usr/lib/python3.7/py_compile.py", line 204, in main
    compile(filename, doraise=True)
  File "/usr/lib/python3.7/py_compile.py", line 169, in compile
    importlib._bootstrap_external._write_atomic(cfile, bytecode, mode)
  File "<frozen importlib._bootstrap_external>", line 112, in _write_atomic
PermissionError: [Errno 13] Permission denied: '/usr/lib/python3.7/__pycache__/__future__.cpython-37.pyc.139921434723696'
dpkg: error processing package python3.7-minimal (--configure):
 installed python3.7-minimal package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 python3.7-minimal
E: Sub-process /usr/bin/dpkg returned an error code (1)
The command '/bin/sh -c apt-get update -y && apt-get install -y python3-dev gcc     build-essential pipenv locales locales-all vim     && rm -rf /var/www/html     && rm -rf /var/lib/apt/lists/*' returned a non-zero code: 100
```

Dockerfile:
```
ARG VERSION_PYTHON="3.10.6-slim-buster"

FROM python:${VERSION_PYTHON}

ENV PYTHONUTNBUFFERED 1

LABEL verion="1.0.0" \
    team="devops"

RUN apt-get update -y && apt-get install -y python3-dev gcc \
    build-essential pipenv locales locales-all vim \
    && rm -rf /var/lib/apt/lists/*

RUN useradd -s /bin/bash check

WORKDIR /app

COPY Pipfile* ./

RUN set -ex && pipenv install --deploy --system

COPY . /app

RUN chown check /app/check_free_space.py

USER  check

CMD [ "/usr/local/bin/python3", "-u", "/app/check_free_space.py" ]
```

Ahmad-Wahid, 2023-05-02:
I have been facing the same issue, and I tried to re-install docker as @amokduke has mentioned. But every time I get this error I should not have to delete and install docker again and again. There might be some other way around.

**Docker file:**

FROM ubuntu:focal
 
ENV DEBIAN_FRONTEND noninteractive
ENV LC_ALL C.UTF-8
ENV LANG C.UTF-8

-# pre-requisites
RUN apt-get update && apt-get install --no-install-recommends -y --upgrade python3 python3-pip git curl gunicorn coinor-cbc postgresql-client && apt-get clean

WORKDIR /app
-# requirements - doing this earlier, so we don't install them each time. Use --no-cache to refresh them.
COPY requirements /app/requirements

-# py dev tooling
RUN python3 -m pip install --no-cache-dir  --upgrade pip && python3 --version && pip3 install --no-cache-dir --upgrade setuptools && pip3 install --no-cache-dir -r requirements/app.txt -r requirements/dev.txt -r requirements/test.txt

-# Copy code and meta/config data
COPY setup.* .flaskenv wsgi.py /app/
COPY flexmeasures/ /app/flexmeasures
#RUN find . | grep -E "(__pycache__|\.pyc|\.pyo$)" | xargs rm -rf
COPY .git/ /app/.git

RUN pip3 install --no-cache-dir .

EXPOSE 5000

CMD [ \
    "gunicorn", \
    "--bind", "0.0.0.0:5000", \
    -# This is set to /tmp by default, but this is part of the Docker overlay filesystem, and can cause stalls.
    -# http://docs.gunicorn.org/en/latest/faq.html#how-do-i-avoid-gunicorn-excessively-blocking-in-os-fchmod
    "--worker-tmp-dir", "/dev/shm", \
    -# Ideally you'd want one worker per container, but we don't want to risk the health check timing out because
    -# another request is taking a long time to complete.
    "--workers", "2", "--threads", "4", \
    "wsgi:application" \
]

Thenerdstation, 2023-05-10:
I also have been having the same issue, and I always have to resort to nuking docker and reinstalling it, but it continues to pop up for me no matter how often I do a clean install. Have any of you found a consistent solution for this?

Ahmad-Wahid, 2023-05-10:
no solution yet.

id2359, 2023-05-12:
On Ubuntu I had a similar permission issue. I was user docker in a snap. I think an upgrade had occurred and switched me to that and I hadn't noticed. I uninstalled docker and then switch to an apt install of docker.io and it worked.

amaysharma, 2023-05-16:
> I have been facing the same issue, and I tried to re-install docker as @amokduke has mentioned. But every time I get this error I should not have to delete and install docker again and again. There might be some other way around.

What's the current status , Are u now been able to do it without reinstalling docker again and again.

amokduke, 2023-05-16:
> What's the current status , Are u now been able to do it without reinstalling docker again and again.
Hi @amaysharma , yes I have resolved the issue. I do not have to reinstall docker every time. I guess you can try double check the steps to reinstall docker… Hope it helps.

hassammustafaa, 2023-08-18:
> > > I have been facing the same issue, and I tried to re-install docker as @amokduke has mentioned. But every time I get this error I should not have to delete and install docker again and again. There might be some other way around.
> > 
> > 
> > What's the current status , Are u now been able to do it without reinstalling docker again and again.
> > Hi @amaysharma , yes I have resolved the issue. I do not have to reinstall docker every time. I guess you can try double check the steps to uninstall and reinstall docker… Hope it helps.

Hi, I tried to follow every steps but whenever server restarts it pop up for some reason?

Hu-Wentao, 2023-08-28:
> On Ubuntu I had a similar permission issue. I was user docker in a snap. I think an upgrade had occurred and switched me to that and I hadn't noticed. I uninstalled docker and then switch to an apt install of docker.io and it worked.

it worked for me:
```bash
sudo apt update
sudo apt install docker.io -y
```

tvup, 2023-09-05:
> I have found the error, it got to do with my docker installation. I resolved the error by:
> 
> 1. Removing docker from my computer COMPLETELY with the steps here: https://www.golinuxcloud.com/ubuntu-uninstall-docker/
> 2. Reinstall docker with the steps here: https://docs.docker.com/engine/install/ubuntu/
> 
> @davidmoyalopez , hope it helps.

Is like firing nukes at sparrows..
anyway, I thought that would surely fix it
and it did.
Thanks.. moving on :)

heitor456, 2023-09-12:
> I have found the error, it got to do with my docker installation. I resolved the error by:
>
> 1. Removing docker from my computer COMPLETELY with the steps here: [...]

It worked for me. 
But I had to do it twice, for different projects. And I'm sure will happen again in other projects.
There must to have a better way to solve this. 
Anyway, thanks for the help.

devLalitx86Repo, 2023-09-15:
The same problem came to me when I tried to run a docker inside an EC2 instance
The Problem: The docker was installed using snap, the problem with this docker is their **storage drivers**
To solve the problem (in my case)
I have added `"storage-driver": "vfs"` to the `/var/snap/docker/current/config/daemon.json`
Also doing that makes all your current containers and images unusable so make sure to close them and restart the docker 
`sudo snap restart docker`

ref: https://stackoverflow.com/questions/72564830/python3-minimal-error-during-pip-installation-in-docker-build-permissionerro

Thistleman, 2023-10-23:
> The same problem came to me when I tried to build a docker inside an EC2 instance The Problem: The docker was installed using snap, the problem with this docker is their **storage drivers** To solve the problem (in my case) I have added `"storage-driver": "vfs"` to the `/var/snap/docker/current/config/daemon.json` Also doing that makes all your current containers and images unusable so make sure to close them and restart the docker `sudo snap restart docker`
> 
> ref: https://stackoverflow.com/questions/72564830/python3-minimal-error-during-pip-installation-in-docker-build-permissionerro

This. This so much, I have to use EC2 since my dev computer is an m1 so it cannot build images usable on ECS. I didn't know EC2 used snap, so I couldn't figure out why or how it stopped working.

Absolute life saver!

sjbailey838, 2023-12-02:
> The same problem came to me when I tried to build a docker inside an EC2 instance The Problem: The docker was installed using snap, the problem with this docker is their **storage drivers** To solve [...]

Confirmed fix for me as well. Thank you!

closed 2024-04-05 by HTRamsey