mavlink/qgroundcontrol #10617: The next version of the QGC UI/UX

DonLakeFlyer, 2023-03-13:
So I'm planning on spending my time moving the QGC UI/UX forward to improve usability. First I'll detail out my opinions on high-level problems with the current QGC:

* In some sense QGC's strengths are also it's current weaknesses:
    * Includes support for as much mavlink spec as possible
    * Includes support for both PX4 Pro and Ardupilot firmware
    * Includes support for multiple specific vehicle types
    * Includes support for multi-os and multi-device (laptop, tablet, mouse, touch)
complexity

* The mavlink spec is missing a ton of stuff which relates to allowing a groundstation to understand a vehicles capabilities and/or hardware elements.
* The target user is somewhere between:
    *  Vehicle developer: Someone who is creating a new vehicle from scratch and needs all the initial setup, configuration and debugging options for testing out something new.
    *  Vehicle consumer: Think of as someone who has a DJI Mavic and uses DJI Go.
*  The target UI is somewhere between focusing on a laptop and a tablet.
*  The target UI is focused on screens which are a bit too large. For example tagetting a 10" screen leaves out Herelink sized displays.

All of the above bleeds through in various places in the UI adding complexity and looking a bit schizophrenic. Some examples:

* Since in general QGC doesn't know enough up front whether the vehicle has a camera or a gimbal:

    * When you are creating a mission you are presented with all sorts of camera and gimbal goop even though your vehicle may have none of those.
    * The Fly View shows a video widget even though you may have no video.
    * The Fly View starts on the map instead of video even though your vechicle DOES have video.
    * The camera widget on the fly view shows you video options, when your vechicle only has picture capture capability.

* Since QGC doesn't know if your vehicle has the sensors needed to perform precions landing it shows you the Precision Land flight mode in the dropdown even though it's not possible. Further cluttering the already massive list of flight modes.

* Initial vehicle setup/configuration is mixed in with every day use settings
    * If you go into Vehicle Setup the majority of stuff in there if for configuring a brand new vehicle from scratch. 
    * For a vehicle consumer this is something you would never need. The vehicle is already set up for you.
    * Even for a DIY vehicle developer it is something you do once and then you go fly. You might re-calibrate the compass every once in a while. But the other stuff you don't go back to.
    * And then in the middle of a huge list of one time vehicle developer setup stuff is a nugget of every daty use settings: The Safety page, which contains things like your RTL height.

Is all of this fixable? Absolutely. I've already started working on it and I'd like to show you all what the starting point looks like. I've posted the video to slack for viewing since it simpler to see it there. I'll also push up a branch soon so people can play around with it to try it out.

DonLakeFlyer, 2023-03-14:
If you want to try a build there are artifacts here: https://github.com/mavlink/qgroundcontrol/pull/10618

bkueng, 2023-03-15:
Hey Don, good to see you again :)

just fyi there's some ongoing effort to dynamically get the list of available flight modes (together with an interface to dynamically add new ones to the FC): https://github.com/mavlink/mavlink/pull/1915

closed 2024-11-30 by HTRamsey