mavlink/qgroundcontrol #10619: Logs download in Analyze Tools but so much slow download rate.

RajanDhiman07, 2023-03-14:
The log downloading speed is too slow on android.

Current Behavior-
Approx 1 MB download take time more than 8 minute and 2-3 kb downloading rate. 

System Information
When posting bug reports, include the following information
QGC build
Flight Controller: X7+
Autopilot (with version):  master.

Screenshot-
[image]

booo, 2023-03-21:
What download speeds do you expect?

How are you connected to the FC? Please share details about your setup.

RajanDhiman07, 2023-03-22:
we are connecting FC with the air unit and we are using RC-H16. and we expect the speed to be 1 MB download complete take time in approx 2 Minutes.

booo, 2023-03-22:
I don't think your expectations are right. E.g. PixRacer + ESP never reaches speeds in the MegaBits/s range. Even with usb connection you don't get that.

Did you get other download speeds with other setups?

RajanDhiman07, 2023-03-22:
Yes received, if I connect with a USB cable speed, is received 200kb per second but if connect with an air unit speed given is an average of 3 to 4 kb per second.

HTRamsey, 2024-05-31:
This isn't QGC specific. Log downloads are just crazy slow over IP

closed 2024-05-31 by HTRamsey