mavlink/qgroundcontrol #10625: NTP server inside QGC

zdanek, 2023-03-17:
I propose to implement NTP server inside QGC.
Source of time would be GPS signal that QGC is aware of. If no GPS is available to QGC computer host, time can be taken from mavlink's GPS.

The purpose of this is to provide source of current time and synchronize all devices that are in the same network. Some sophisticated setups can contain a few devices at base station side.

My use case is that we're using Nvidia Jetson but there's no time source in local net. Same could be when using RPi - both don't have RTC (battery supported clock). Of course we have tablet  and/or PC with QGC but it would require installing an external NTP server to provide times ource from PC's clock. NTP itself it very simple and could be inside QGC.

**I can develop this feature myself and make PR to QGC. It would be as an optional service that means it could be enabled in the settings. Nothing mandatory**

What do you think about this feature?

zdanek, 2023-03-21:
Oh, thanks for the tip. I'll check that. 
My companion computer (RPi) also suffers from invalid date and time. I though about feeding system time from mavlink (gps from pixhawk). Maybe gpsd driver you mentioned would be good idea. I will investigate that and come back with my thoughts.

booo, 2023-03-21:
I can share some snippets of python code we used to set the time based on mavlink gps message.

zdanek, 2023-03-21:
That would be great. Thank you. 
Current time is crucial for example for good logging to check what is happening to the system broadly - in ground station and in the same time on companion computer. If time is random, one will never get what's going on :)

booo, 2023-03-21:
And another solution with pure mavlink messages:

https://gitlab.com/searchwing/development/payloads/searchwing-pi/-/blame/master/src/zmqDrivers/droneState.py#L57

closed 2026-10-04 by github-actions[bot]