mavlink/qgroundcontrol #10627: Problem running RTSP on Linux

fraviofii, 2023-03-17:
---

## Expected Behavior

RTSP Video should appear on the QGroundControl screen.

## Current Behavior

RTSP Video not showing up for my specific RTSP Server.

## Steps to Reproduce:

1. Setup Video Stream from RTSP Server
2. Point to the RTSP server (in my case, another computer in my LAN)
3. Go back, to save and start the configuration

## System Information

- Operating System: Linux Debian Jessie
- QGC Version: lastest in git
- QGC build: master
- Flight Controller: Ardupilot
- Autopilot (with version): Ardupilot Copter 3.6.1

## Detailed Description

I'm having problems to connect to my custom RTSP server using QGroundControl. When using the Bunny test, it works.

This problem happens only in Linux. It works when built for Windows and Android.

I can get the communication using VLC. From there, I got the following information:

```
Codec: H264 - MPEG-4 AVC (part 10) (h264)
Type: Video
Video resolution: 1280x720
Buffer dimensions: 1280x738
Frame rate: 30
Decoded format: Planar 4:2:0 YUV
Orientation: Top Left
Chrome location: Left
```

The error that appears on QT Creator console is:

```
VideoReceiverLog: Gstreamer error: Internal data stream error
```

I could track where this error happens, and I found that it is called on `_onBusMessage` (`GstVideoReceiver.cc` line `1036`). This function is a signal handler initalized on `start`  method.

I tried to execute the commands using Gstreamer directly. And it works with the following command:

```
gst-launch-1.0 rtspsrc location=rtsp://192.168.100.137:34554 ! application/x-rtp ! rtpjitterbuffer ! rtph264depay ! avdec_h264 ! videoconvert ! autovideosink
```

This command is extracted from [VideoReceiver README](https://github.com/mavlink/qgroundcontrol/blob/master/src/VideoReceiver/README.md).

I realized that the method *_addVideoSink* is never called. Is this signal handler call some kind of problem?

How can this problem be solved?

## Log Files and Screenshots

Please help me on how to provide additional information.

zdanek, 2023-03-19:
Hi.
Enable video logs with option added to QGC executable
`--logging:VideoAllLog`
and let's see what happened.

fraviofii, 2023-03-20:
Hi @zdanek ,

Here is the output from the log using the command line argument that you advised:

```
QML debugging is enabled. Only use this in a safe environment.
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
Settings location "/home/debian/.config/QGroundControl.org/QGroundControl Daily.ini" Is writable?: true
Filter rules "*Log.debug=false\nVideoAllLog.debug=true\nVideoManagerLog.debug=true\nVideoReceiverLog.debug=true\nGStreamerLog.debug=true\nGStreamerAPILog.debug=true\nqt.qml.connections=false"
System reported locale: QLocale(English, Latin, United States) ; Name "en_US" ; Preffered (used in maps):  "en-US"
serialnmea: No known GPS device found. Specify the COM port via QT_NMEA_SERIAL_PORT.
GStreamerLog: Failed to change ranking of feature. Featuer does not exist: bcmdec
VideoManagerLog: New Video Source: "RTSP Video Stream"
VideoManagerLog: Testing configuration for RTSP Stream: "rtsp://192.168.100.118:34554"
MAVLinkLogManagerLog: MAVLink logs directory: "/home/debian/Documents/QGroundControl Daily/Logs"
VideoReceiverLog: Starting "rtsp://192.168.100.118:34554" , buffer 0
VideoReceiverLog: Started "rtsp://192.168.100.118:34554"
Map Cache in: "/home/debian/.cache/QGCMapCache300" / "qgcMapCache.db"
VideoReceiverLog: Linking pad start
VideoReceiverLog: done
VideoReceiverLog: Linking pad end
VideoReceiverLog: Streaming started "rtsp://192.168.100.118:34554"
setCurrentPlanViewSeqNum
setCurrentPlanViewSeqNum
VideoManagerLog: Video 0 Start complete, status:  VideoReceiver::STATUS_OK
qt.positioning.geoclue2: Unable to start the client: "org.freedesktop.DBus.Error.AccessDenied" "Geolocation disabled for UID 1000"
QGCPositionManager error 0
VideoReceiverLog: Starting decoding "rtsp://192.168.100.118:34554"
VideoReceiverLog: Decoding started "rtsp://192.168.100.118:34554"
"v4.2.0"
VideoReceiverLog: Stream timeout, no frames for  9  "rtsp://192.168.100.118:34554"
VideoReceiverLog: Stopping "rtsp://192.168.100.118:34554"
VideoReceiverLog: Streaming stopped "rtsp://192.168.100.118:34554"
VideoReceiverLog: Stopped "rtsp://192.168.100.118:34554"
VideoManagerLog: Video 0 Stop complete, status:  VideoReceiver::STATUS_OK
VideoReceiverLog: Starting "rtsp://192.168.100.118:34554" , buffer 0
VideoReceiverLog: Started "rtsp://192.168.100.118:34554"
VideoManagerLog: Video 0 Start complete, status:  VideoReceiver::STATUS_OK
VideoManagerLog: Video 0 start decoding
VideoReceiverLog: Starting decoding "rtsp://192.168.100.118:34554"
VideoReceiverLog: Linking pad start
VideoReceiverLog: done
VideoReceiverLog: Linking pad end
VideoReceiverLog: Streaming started "rtsp://192.168.100.118:34554"
VideoReceiverLog: Decoding started "rtsp://192.168.100.118:34554"
VideoReceiverLog: GStreamer debug:  ../libs/gst/base/gstbasesrc.c(3127): gst_base_src_loop (): /GstPipeline:receiver/GstBin:sourcebin/GstRTSPSrc:source/GstUDPSrc:udpsrc2:
streaming stopped, reason not-negotiated (-4)
VideoReceiverLog: Fii GStreamer error: Internal data stream error.
VideoReceiverLog: Stopping because of error
VideoReceiverLog: Stopping "rtsp://192.168.100.118:34554"
VideoReceiverLog: Streaming stopped "rtsp://192.168.100.118:34554"
VideoReceiverLog: Stopped "rtsp://192.168.100.118:34554"
VideoManagerLog: Video 0 Stop complete, status:  VideoReceiver::STATUS_OK
VideoReceiverLog: Starting "rtsp://192.168.100.118:34554" , buffer 0
VideoReceiverLog: Started "rtsp://192.168.100.118:34554"
VideoManagerLog: Video 0 Start complete, status:  VideoReceiver::STATUS_OK
VideoManagerLog: Video 0 start decoding
VideoReceiverLog: Starting decoding "rtsp://192.168.100.118:34554"
VideoReceiverLog: Linking pad start
VideoReceiverLog: done
VideoReceiverLog: Linking pad end
VideoReceiverLog: Streaming started "rtsp://192.168.100.118:34554"
VideoReceiverLog: Decoding started "rtsp://192.168.100.118:34554"
VideoReceiverLog: Stream timeout, no frames for  9  "rtsp://192.168.100.118:34554"
VideoReceiverLog: Stopping "rtsp://192.168.100.118:34554"
VideoReceiverLog: Streaming stopped "rtsp://192.168.100.118:34554"
VideoReceiverLog: Stopped "rtsp://192.168.100.118:34554"
VideoManagerLog: Video 0 Stop complete, status:  VideoReceiver::STATUS_OK
VideoReceiverLog: Starting "rtsp://192.168.100.118:34554" , buffer 0
VideoReceiverLog: Started "rtsp://192.168.100.118:34554"
VideoManagerLog: Video 0 Start complete, status:  VideoReceiver::STATUS_OK
VideoManagerLog: Video 0 start decoding
VideoReceiverLog: Starting decoding "rtsp://192.168.100.118:34554"
VideoReceiverLog: Linking pad start
VideoReceiverLog: done
VideoReceiverLog: Linking pad end
VideoReceiverLog: Streaming started "rtsp://192.168.100.118:34554"
VideoReceiverLog: Decoding started "rtsp://192.168.100.118:34554"
```

Thank you for this advice. The problem is here: `VideoReceiverLog: GStreamer debug:  ../libs/gst/base/gstbasesrc.c(3127): gst_base_src_loop ()::/GstPipeline:receiver/GstBin:sourcebin/GstRTSPSrc:source/GstUDPSrc:udpsrc2:streaming stopped, reason not-negotiated (-4)`

Looking at the issues, I found some [similar issue](https://github.com/mavlink/qgroundcontrol/issues/10542). I tried to add the `videoconvert` to the GStreamer pipeline, but I could not find the correct place to do that.

In addition, I don't know if it is the correct solution.

Can you help me with that?

zdanek, 2023-03-22:
Not enough.
Set `GST_DEBUG=3` env variable before running QGC. If this will not show any errors, higher debug value 4,5 or even 6 might be needed but it will generate big output so please check if lower debug values would be sufficient.

fraviofii, 2023-03-22:
Hello @zdanek ,

I set `GST_DEBUG=5` during the execution of QGC.

Here is the output until the first occurrence of the error:

```
0:00:13.187532716  3226 0x7fff78028980 DEBUG               GST_PADS gstpad.c:4252:gst_pad_peer_query:<sink:proxypad9> peer query 0x7fff6000b800 (accept-caps)
0:00:13.187544483  3226 0x7fff78028980 DEBUG               GST_PADS gstpad.c:4125:gst_pad_query:<parsebin0:sink> doing query 0x7fff6000b800 (accept-caps)
0:00:13.187549843  3226 0x7fff78028980 DEBUG               GST_PADS gstpad.c:3195:gst_pad_query_accept_caps_default:<parsebin0:sink> query accept-caps accept-caps query: 0x7fff6000b800, GstQueryAcceptCaps, caps=(GstCaps)"video/x-h264\,\ alignment\=\(string\)au\,\ stream-format\=\(string\)avc\,\ parsed\=\(boolean\)true\,\ width\=\(int\)1280\,\ height\=\(int\)720\,\ framerate\=\(fraction\)30/1\,\ chroma-format\=\(string\)4:2:0\,\ bit-depth-luma\=\(uint\)8\,\ bit-depth-chroma\=\(uint\)8\,\ profile\=\(string\)high\,\ level\=\(string\)4", result=(boolean)false;
0:00:13.187552133  3226 0x7fff78028980 DEBUG               GST_PADS gstutils.c:2782:gst_pad_proxy_query_accept_caps:<parsebin0:sink> proxying accept caps query for parsebin0:sink
0:00:13.187553470  3226 0x7fffa0018640 DEBUG         GST_SCHEDULING gstpad.c:4401:gst_pad_chain_data_unchecked:<manager:recv_rtp_sink_0> called chainfunction &gst_proxy_pad_chain_default with buffer 0x7fff7804ed80, returned ok
0:00:13.187554589  3226 0x7fff78028980 DEBUG               GST_PADS gstpad.c:4252:gst_pad_peer_query:<sink:proxypad10> peer query 0x7fff6000b800 (accept-caps)
0:00:13.187562826  3226 0x7fff78028980 DEBUG               GST_PADS gstpad.c:4125:gst_pad_query:<typefind:sink> doing query 0x7fff6000b800 (accept-caps)
0:00:13.187565688  3226 0x7fffa0018640 DEBUG               GST_PADS gstpad.c:1673:gst_pad_check_reconfigure:<udpsrc1:src> remove RECONFIGURE flag
0:00:13.187567871  3226 0x7fff78028980 DEBUG               GST_PADS gstpad.c:3195:gst_pad_query_accept_caps_default:<typefind:sink> query accept-caps accept-caps query: 0x7fff6000b800, GstQueryAcceptCaps, caps=(GstCaps)"video/x-h264\,\ alignment\=\(string\)au\,\ stream-format\=\(string\)avc\,\ parsed\=\(boolean\)true\,\ width\=\(int\)1280\,\ height\=\(int\)720\,\ framerate\=\(fraction\)30/1\,\ chroma-format\=\(string\)4:2:0\,\ bit-depth-luma\=\(uint\)8\,\ bit-depth-chroma\=\(uint\)8\,\ profile\=\(string\)high\,\ level\=\(string\)4", result=(boolean)false;
0:00:13.187576310  3226 0x7fff78028980 DEBUG        GST_PERFORMANCE gstpad.c:3212:gst_pad_query_accept_caps_default:<typefind:sink> fallback ACCEPT_CAPS query, consider implementing a specialized version
0:00:13.187576186  3226 0x7fffa0018640 DEBUG                basesrc gstbasesrc.c:3439:gst_base_src_negotiate_unlocked:<udpsrc1> starting negotiation
0:00:13.187580821  3226 0x7fff78028980 DEBUG               GST_CAPS gstutils.c:3102:gst_pad_query_caps:<typefind:sink> get pad caps with filter video/x-h264, alignment=(string)au, stream-format=(string)avc, parsed=(boolean)true, width=(int)1280, height=(int)720, framerate=(fraction)30/1, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, profile=(string)high, level=(string)4
0:00:13.187581931  3226 0x7fffa0018640 DEBUG               GST_CAPS gstutils.c:3102:gst_pad_query_caps:<udpsrc1:src> get pad caps with filter (NULL)
0:00:13.187585152  3226 0x7fff78028980 DEBUG                  query gstquery.c:679:gst_query_new_custom: creating new query 0x7fff6000b770 caps
0:00:13.187587853  3226 0x7fff78028980 DEBUG               GST_PADS gstpad.c:4125:gst_pad_query:<typefind:sink> doing query 0x7fff6000b770 (caps)
0:00:13.187589004  3226 0x7fffa0018640 DEBUG                  query gstquery.c:679:gst_query_new_custom: creating new query 0x7fff7804f4f0 caps
0:00:13.187593367  3226 0x7fff78028980 DEBUG               GST_CAPS gstpad.c:3254:gst_pad_query_caps_default:<typefind:sink> query caps caps query: 0x7fff6000b770, GstQueryCaps, filter=(GstCaps)"video/x-h264\,\ alignment\=\(string\)au\,\ stream-format\=\(string\)avc\,\ parsed\=\(boolean\)true\,\ width\=\(int\)1280\,\ height\=\(int\)720\,\ framerate\=\(fraction\)30/1\,\ chroma-format\=\(string\)4:2:0\,\ bit-depth-luma\=\(uint\)8\,\ bit-depth-chroma\=\(uint\)8\,\ profile\=\(string\)high\,\ level\=\(string\)4", caps=(GstCaps)"NULL";
0:00:13.187594020  3226 0x7fffa0018640 DEBUG               GST_PADS gstpad.c:4125:gst_pad_query:<udpsrc1:src> doing query 0x7fff7804f4f0 (caps)
0:00:13.187595402  3226 0x7fff78028980 DEBUG               GST_CAPS gstpad.c:3278:gst_pad_query_caps_default:<typefind:sink> trying pad template caps
0:00:13.187601875  3226 0x7fffa0018640 DEBUG               GST_PADS gstpad.c:3887:do_probe_callbacks:<udpsrc1:src> data is passed
0:00:13.187606820  3226 0x7fff78028980 DEBUG               GST_CAPS gstpad.c:3300:gst_pad_query_caps_default:<typefind:sink> using caps 0x7fffdc00acf0 ANY with filter 0x7fff6000b850 video/x-h264, alignment=(string)au, stream-format=(string)avc, parsed=(boolean)true, width=(int)1280, height=(int)720, framerate=(fraction)30/1, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, profile=(string)high, level=(string)4
0:00:13.187611234  3226 0x7fff78028980 DEBUG               GST_CAPS gstpad.c:3304:gst_pad_query_caps_default:<typefind:sink> result 0x7fff6000b850 video/x-h264, alignment=(string)au, stream-format=(string)avc, parsed=(boolean)true, width=(int)1280, height=(int)720, framerate=(fraction)30/1, chroma-format=(string)4:2:0,  [... cut ...]
0:00:13.236648837  3226 0x7fffa0018640 DEBUG               GST_PADS gstpad.c:3986:push_sticky:<recv_rtp_sink_0:proxypad1> event segment was already received
VideoReceiverLog: Stopping because of error
0:00:13.236651789  3226 0x7fffa0018640 DEBUG              GST_EVENT gstpad.c:5774:gst_pad_send_event_unchecked:<rtpsession0:recv_rtp_sink> have event type eos event: 0x7fff7804c240, time 99:99:99.999999999, seq-num 127, (NULL)
0:00:13.236667359  3226 0x7fffa0018640 DEBUG             rtpsession gstrtpsession.c:1734:gst_rtp_session_event_recv_rtp_sink:<rtpsession0> received event eos
VideoReceiverLog: Stopping "rtsp://192.168.100.118:34554"
```

fraviofii, 2023-05-18:
It took me some time to finally find what is the error that is happening:

```
GStreamerAPILog: (NULL) No start code is found.
GStreamerAPILog: (NULL) Error splitting the input into NAL units.
```

Is there a way to solve, or workaround, this issue?