mavlink/qgroundcontrol #10632: Joystick support overhaul

RommanirsMer, 2023-03-21:
### Tell us a bit about the feature:

The issue is the current system is only allowing to bind joystick that handle at least 4 axis in one device.
However most of current modern consumer HOTAS, (also HOSAS,etc  ...) system are now using separate peripheral such as Thrustmaster warthog, etc... to handle various tasks without removing hands of controls.  

If its a stick it will handle up to 3 axis and throttle up to 2 while ignoring aditional input such as ministick, button, castle switch, etc...

Due to that observation the current joystick implementation feel a bit outdated while looking at new peripheral options that are in my opinion way more efficient to use for manual handling of sensors/vehicles to me than regular radio system.

### What problem does it solve?:

An overhaul of joysticks handling to map X number of peripherals inputs to fulfil X tasks or use X axis will improve vehicle manual handling. Where regular radio system will thoses number of inputs for being a smaller packages.

For exemple: 
It will allow to fly a drone while operating a gimbal camera with a mini joystick or multi way switch, and zoom with dedicated button or axis with both hands. 

Where radio emitter will force an operator to drop flights controls, degrade sticks handling, change mode to operate other controls.

It will also favorise the development of sepecifc controls setup for a vehicle.

### What flight stacks must it work with? (All, PX4, ArduPilot)
This overhaul will benifit every flight stack 

### Relevant vehicle types? (All, multirotor, fixed-wing, VTOL, submarine, etc.)
This overhaul will benefit every vehicule

### Are there other systems that have this feature?
QGC is my only GCS option, i don't really know

zdanek, 2023-03-21:
I'm interested in this feature.

booo, 2023-03-21:
Are you able to contribute such a overhaul some time soon?

The resources of our small developer community are quite limited. Don't expect to many complex feature requests to be dealt with. We are happy to give you directions if you want to tackle the problem.

RommanirsMer, 2023-03-21:
I'have planned to dig a bit on the subject indeed

but i need to finished my vehicule for a keynote (school stuff) which is quite time consuming.

There was a plan for it in my workgroup at the beggining of my project but they were lost through QGC source code 
So i'm going to do it myselfonce i'have settled my vehicule

AGSBurei, 2023-05-23:
Ok, I'm done with the school stuff.
I'm going to dive into the subject, which files should i give a look ? (expected /src/Joystick)

AGSBurei, 2023-05-24:
I'have made a quick photoshop sketch on which kind of change i would like to make before reverse engeenering the code 
what do you guys think about it ? 
[image]
[image]
[image]

APS-Space, 2023-06-02:
Please do continue to work on this feature, I have just found this issue today, and it would be great if this came into fruition sometime in the near future. Whilst I only really know python, and have school to deal with, I am not really able to make a direct contribution to this. But, if of any assistance I could help with testing and debugging of the system.

Many thanks, APS-Space

AGSBurei, 2023-06-02:
> I just want to let you know that we would appreciate your contribution a lot. I have no real prior experience with the joystick setup and can't help much but please commit early and often and let us know about your progress. If you have any concrete questions please post them here. Maybe someone with prior experience can help out.
> 
> I think your mock up looks good and makes sense.

Well its an oportunity for me to learn new stuff and i kind of need it also

AGSBurei, 2023-08-01:
[image]
Ok i'have find it
i had to set an alias for the QML on qgroundcontrol.qrc to make it work