mavlink/qgroundcontrol #10633: [BUG] Custom build fails under Windows, QT creator

zdanek, 2023-03-21:
## Expected Behavior
I'm Ubuntu user, but I need to build Windows version, so I installed Qt and MS Visual studio compiler. Rest is simple - load project under QT Creator, setup project, build.
So I expect to be able to build example custom build under Windows with QT Creator.

## Current Behavior
Building offical release works. 
Building custom version by renaming `custom-example` to `custom` and running build gives build error. 

## Steps to Reproduce:
Checkout current master `742a3e4`
Rename `custom-example` to `custom`
Build as usual.

Without renaming (e.i. enabling custom build) all works great.

## System Information
- Operating System: Windows 10
- QGC Version: master, `742a3e4` (latest stable is 4.2.6)
- QGC build: self-built from source `742a3e4`

## Log Files and Screenshots
```
07:43:37: Running steps for project qgroundcontrol...
....
Could Not Find C:\build-qgroundcontrol-Desktop_Qt_5_15_2_MSVC2015_64bit-Debug\CustomQGroundControl_pch.pch
Could Not Find C:\build-qgroundcontrol-Desktop_Qt_5_15_2_MSVC2015_64bit-Debug\staging\CustomQGroundControl.ilk
	del CustomQGroundControl_resource.res
Could Not Find C:\build-qgroundcontrol-Desktop_Qt_5_15_2_MSVC2015_64bit-Debug\CustomQGroundControl_resource.res
07:43:39: The process "C:\Qt\Tools\QtCreator\bin\jom\jom.exe" exited normally.
07:43:39: Starting: "C:\Qt\5.15.2\msvc2015_64\bin\qmake.exe" C:\qgc-gh\qgroundcontrol.pro -spec win32-msvc "CONFIG+=debug" "CONFIG+=qml_debug"
Project MESSAGE: Qt version 5.15.2
Project MESSAGE: Windows build
Project MESSAGE: Daily Build
Project MESSAGE: GIT_DESCRIBE v4.2.0-322-g742a3e4fb
Project MESSAGE: QGroundControl APP_VERSION_STR VERSION v4.2.0-322-g742a3e4fb 4.2.0.322
Project MESSAGE: Debug flavor
Project MESSAGE: Found custom build
Project MESSAGE: Adding Custom Plugin
Project MESSAGE: Custom QGC Version: 0.0.7129
Project MESSAGE: Disable APM Plugin
Project MESSAGE: Disable APM Plugin Factory
Project MESSAGE: Disable PX4 Plugin Factory
Project MESSAGE: Skipping support for Pairing
Project MESSAGE: Disable APM MAVLink support
Project MESSAGE: Using MAVLink dialect 'common'.
Project MESSAGE: Skipping support for Zeroconf (unsupported platform)
Project MESSAGE: Including support for AirMap
Project MESSAGE: Using custom qgroundcontrol.qrc
Project MESSAGE: Using custom qgcresources.qrc
Project MESSAGE: Using custom InstrumentValueIcons.qrc
Project MESSAGE: Taisync disabled
Project MESSAGE: Microhard disabled
Project MESSAGE: Skipping support for video streaming (GStreamer libraries not installed)
Project MESSAGE: Installation instructions here: https://github.com/mavlink/qgroundcontrol/blob/master/src/VideoReceiver/README.md
Project MESSAGE: This project is using private headers and will therefore be tied to this specific Qt module build version.
Project MESSAGE: Running this project against other versions of the Qt modules may crash at any arbitrary point.
Project MESSAGE: This is not a bug, but a result of using Qt internals. You have been warned!
07:43:47: The process "C:\Qt\5.15.2\msvc2015_64\bin\qmake.exe" exited normally.
07:43:47: Starting: "C:\Qt\Tools\QtCreator\bin\jom\jom.exe" -f C:/build-qgroundcontrol-Desktop_Qt_5_15_2_MSVC2015_64bit-Debug/Makefile qmake_all

jom 1.1.3 - empower your cores

07:43:48: The process "C:\Qt\Tools\QtCreator\bin\jom\jom.exe" exited normally.
07:43:48: Starting: "C:\Qt\Tools\QtCreator\bin\jom\jom.exe" 
Error: dependent 'C:\qgc-gh\src\QmlControls\QGCViewDialog.qml' does not exist.
07:43:49: The process "C:\Qt\Tools\QtCreator\bin\jom\jom.exe" exited with code 2.
Error while building/deploying project qgroundcontrol (kit: Desktop Qt 5.15.2 MSVC2015 64bit)
When executing step "Make"
07:43:49: Elapsed time: 00:12.
```

booo, 2023-03-21:
Please share the fixes you have so far.

Maybe we should add the custom builds to the CI system? Otherwise I guess they get neglected.

zdanek, 2023-03-21:
I've found the cause. I'll provide PR soon.

It's good idea to od custom build example to CI.

I've also prepared quite broad fix to cmake build files to support custom
builds with cmake. Currently not supported. Cmake ignores custom dir.

zdanek, 2023-03-22:
I have another problem. I'm linux guy and all works great there. Even with CLion as IDE.
Now I have to build QGC for Windows 10. I managed to build it and run at particular windows computer.
But when I copy exe file to other Windows lots of dlls are missing. Copying them by hand makes it start with another strange error  0xc00007 somehow connected to *not* installed runtimes. I will dig into that but my basic question to the community is
**how you work with development of QGC under Windows?**
What IDE do you use? How do you build it and redistribute if needed? Windows makes me confused at every step. Documentation states about QT creator and I'm using it but for me it's so different than CLion and even MS visual studio that I used years back so I'm constantly confused :)

Any tips, comments, suggestions would be very appreciated as I *must* build custom QGC for windows :)

TomSmooth, 2023-05-13:
Hi, have this issue aswell, I have missing QML files which I copied from and older version but now it is looking for files within the custom folder that don't exist? Thanks

ybabs, 2023-07-11:
Hi @zdanek , could you please share what changes you made to get v4.2 build with the custom-example? We're having the same issues here as well.

changx03, 2023-10-31:
The same issue also exist on Linux machine (Ubuntu 22.04). 
The build errors are: `mavlink_rangefinder_t` and `MAVLINK_MSG_ID_RANGEFINDER` are undefined in `Vehicle.cc`. 
Both variables are defined in [ardupilotmega/mavlink_msg_rangefinder.h](https://github.com/mavlink/c_library_v2/blob/a792c5208055337d69efff769a14847620355f83/ardupilotmega/mavlink_msg_rangefinder.h#L10)
This is file correctly downloaded in the local directory (No issue with recursive pull submodule here), but somehow it's not recognized by the compiler.
The code will compile after commenting out Line 725-727 and 1001-1006 in `Vehicle.cc`.
This error persists in both `main` and `Stable_V4.2` branches.

closed 2024-11-30 by HTRamsey