mavlink/qgroundcontrol #10638: Video decode error on Nvidia Jetson device.

Jcwscience, 2023-03-25:
## Expected Behavior
Display video stream

## Current Behavior
Blank screen

## Steps to Reproduce:
Compile on Nvidia Jetson device,
Select any video decoder,
Transmit RTP stream to local port.

## System Information
When posting bug reports, include the following information
- Operating System: Ubuntu 20.04.6 LTS Nvidia Jetson, arm64
- QGC Version: Main Branch
- QGC build: self-built from source
- Flight Controller: n/a
- Autopilot (with version): n/a

## Detailed Description
I receive these errors,
VideoReceiverLog: Unable to link video sink
VideoReceiverLog: _addVideoSink() failed

I believe It is because Nvidia Jetson gstreamer does not use the nvh264dec pipeline element but instead uses
h264parse ! nvv4l2decoder ! nvvidconv

For instance this gstreamer command plays an rtp stream on the device
gst-launch-1.0` -v udpsrc uri=udp://0.0.0.0:5000 caps="application/x-rtp,media=(string)video,clock-rate=(int)90000,encoding-name=(string)H264" ! rtph264depay ! h264parse ! nvv4l2decoder ! nvvidconv ! autovideosink

Is there a way to modify GStreamer.cc and GstVideoReceiver.cc to make this work? I will be attempting to work this out myself over time, but any help or just suggestions in the right direction would be most appreciated!

zdanek, 2023-03-25:
Did you build QGC jetson? Other features work as expected?

Jcwscience, 2023-03-25:
@zdanek I just built the main qgc branch on my device. Everything else seems to work fine. It took a lot of doing to get qt5 compiled too but in the end the video feed is the only thing not working

zdanek, 2023-03-25:
@Jcwscience this is very interesting that I've built QGC on Jetson. Was is on Nano? Stock Ubuntu is 18.xx 
I found 20.04 as unofficial built but most of nv* plugins don't work there.
Can you please elaborate more about this?

Regarding your problem, please generate some logs with:
`qgroundcontrol  --logging:VideoAllLog,GStreamerLog,GStreamerAPILog`

Jcwscience, 2023-03-25:
@zdanek 

GStreamerLog: Failed to change ranking of feature. Featuer does not exist: bcmdec

GStreamerLog: Failed to change ranking of feature. Featuer does not exist: nvh265dec

GStreamerLog: Failed to change ranking of feature. Featuer does not exist: nvh265sldec

GStreamerLog: Failed to change ranking of feature. Featuer does not exist: nvh264dec

GStreamerLog: Failed to change ranking of feature. Featuer does not exist: nvh264sldec

VideoManagerLog: New Video Source: "UDP h.264 Video Stream"

VideoManagerLog: Testing configuration for UDP Stream: 5000

VideoReceiverLog: Starting "udp://0.0.0.0:5000" , buffer 0

VideoReceiverLog: Started "udp://0.0.0.0:5000"

VideoReceiverLog: Streaming started "udp://0.0.0.0:5000"

VideoManagerLog: Video 0 Start complete, status:  VideoReceiver::STATUS_OK

VideoReceiverLog: Starting decoding "udp://0.0.0.0:5000"

VideoReceiverLog: Decoding started "udp://0.0.0.0:5000"

VideoReceiverLog: _onNewDecoderPad "udp://0.0.0.0:5000"

VideoReceiverLog: Unable to link video sink

VideoReceiverLog: _addVideoSink() failed

Opening in BLOCKING MODE 
NvMMLiteOpen : Block : BlockType = 261 
NVMEDIA: Reading vendor.tegra.display-size : status: 6 
NvMMLiteBlockCreate : Block : BlockType = 261

zdanek, 2023-03-27:
@Jcwscience seems that Jetson needs own pipeline. 
I can fix it but I need to know 
1) which Jetson you use
2) where did you get Ubuntu 20.04 image as for Nano (which I have) it's shipped with Ubuntu 18

I need that info and maybe some help with building QGC so I can have proper setup. Then fixing it should be easy.

Jcwscience, 2023-03-27:
@zdanek Ah, ok so I have the AGX Xavier module which now ships with ubuntu 20. They stopped updating the nano modules on 18 I believe.

As for building it, I first compiled qt5 from source, then I followed the compile instructions for Linux. I had to disable airmap since it gave an error also.

Jcwscience, 2023-03-27:
I used the steps in the 3rd comment of this thread to build qt5
https://forums.developer.nvidia.com/t/pyside2-qt-for-python-installation-on-jetson-xavier/160796/4

Jcwscience, 2023-04-06:
@zdanek Any progress?

zdanek, 2023-04-07:
@Jcwscience not really, as I have Jetson nano with 18.04 stock image or hacked 20.04 made by independent people, but without some of nv* plugins to gstreamer, as it's said (on nvidia forums) that gstreamer 1.16 doesn't support some functions and thus some nv* can't be compiled, they say.

So currently I'm afraid I cannot reproduce your problems :-/

lida2003, 2024-11-25:
@Jcwscience @zdanek  Any progress? I have the same issue here on Jetson Orin on L4T 35.6 which is ubuntu 20.04

closed 2026-10-05 by github-actions[bot]