mavlink/qgroundcontrol #10640: Add mechanism to enable ROI features for ArduPilot vehicles

sypaq-nexton, 2023-03-27:
By default, ArduPilot vehicles don't have the [`ROIModeCapability`](https://github.com/mavlink/qgroundcontrol/blob/master/src/FirmwarePlugin/FirmwarePlugin.h#L54) enabled. This means that Region of Interest (ROI) related features can't be used. The two key ROI features are:
* click on the map in the Fly View and send a "ROI at Location" command; and
* add an ROI as part of a Mission Plan.

It would be nice if there was a way to enable these features for ArduPilot vehicles that support ROIs.

My understanding is that `ROIModeCapability` is disabled in the ArduPilot plugin because there is no simple way to tell if a specific build of ArduPilot firmware supports the ROI features. However, there are a couple of ways this _could_ be detected:
1. Add a user setting to force enable the ROI features in QGC;
2. Monitor the MAVLink messages and enable the ROI capability if certain MAVLink messages are received;
3. Check the vehicle parameters to see if a Mount/Gimbal is enabled (e.g. if `MNT_TYPE` is non-zero).

sypaq-nexton, 2023-03-27:
> 2. Monitor the MAVLink messages and enable the ROI capability if certain MAVLink messages are received;

The two problems with this method are:
1. The Mount/Gimbal MAVLink messages sent by ArduPilot are different based on what version of ArduPilot is being used; and
2. Some of the MAVLink messages sent are part of the [`ardupilotmega`](https://mavlink.io/en/messages/ardupilotmega.html]) MAVLink dialect.

DonLakeFlyer, 2023-03-27:
> My understanding is that `ROIModeCapability` is disabled in the ArduPilot plugin because there is no simple way to tell if a specific build of ArduPilot firmware supports the ROI features.

I seem to remember from what back when it was more than just that. That said is it really the end of the world if we just turn it on for ArduPilot all the time? Then it just fails if it won't work.

sypaq-nexton, 2023-03-27:
> is it really the end of the world if we just turn it on for ArduPilot all the time?

Well, that would be a lot easier :)

I've submitted a second PR: #10643

sypaq-nexton, 2023-05-01:
Closing, as #10643 has been merged.

closed 2023-05-01 by sypaq-nexton