mavlink/qgroundcontrol #10644: QMultiMedia

JoshikaNetha, 2023-03-28:
In QGC, why can't we use QMultiMedia, MediaPlayer and VideoOutput. We have GStreamer, but I don't want to use GStreamer, I would like to use QMultiMedia. 

``` 

` Rectangle{ 

        id:                              video 

        anchors.leftMargin:     _toolsMargin + parentToolInsets.leftEdgeCenterInset 

        anchors.topMargin:      _toolsMargin + parentToolInsets.topEdgeLeftInset 

        anchors.left:           parent.left 

        anchors.top:            toolStrip.bottom 

        anchors.bottom:      parent.bottom 

        anchors.bottomMargin: _toolsMargin + parentToolInsets.topEdgeLeftInset 

        z:                      QGroundControl.zOrderWidgets+2 

        height:parent.height 

        color:"white" 

        width:parent.width* 0.3 

        MediaPlayer { 

            id: player1 

            source: VideoPath 

            autoPlay: true 

        } 

        VideoOutput { 

            id: videoOutput1 

            anchors.fill: parent 

            source: player1 

        } 

        Button{ 

            id:clic 

            text:"play" 

            onClicked: { 

                player1.play() 

            } 

        } 

    }` 

``` 

  

  

I have tried above code in FlyVIewWidgetLayer , but its not working. 

Thanks in Advance.

DonLakeFlyer, 2023-03-28:
I think it's the difference of playing back a file versus streaming video over some sort of connection.

harshmittal2210, 2023-03-30:
I tried using the following code once in QGC but it was giving me an error.

```qml
import QtMultimedia 5.15

Rectangle{
       MediaPlayer { 
            id: player
            source: "rtsp://localhost:8554/main.h264" 
            autoPlay: true 
        } 

        VideoOutput { 
            id: videoOutput
            anchors.fill: parent 
            source: player
        } 
}
```

`ERROR: defaultServiceProvider::requestService(): no service found for - "org.qt-project.qt.mediaplayer"`

But if I add it in a new project in Qt and build then no issue and the stream is coming fine.

closed 2024-04-05 by HTRamsey