mavlink/qgroundcontrol #10647: Can't flash ArduPilot, ChibiOS, Multirotor stack to Pixhawk 2.4.8

rtalans, 2023-03-29:
## Expected Behavior
Successful flash of ArduPilot, ChibiOS, Multirotor stack to Pixhawk 2.4.8

## Current Behavior
After following instructions and clicking the OK button and waiting quite a long time, nothing happens. No progress, no messages. QGC is not hung as I can cancel the operation.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior

1. Start **QGC v4.2.4**
2. Connect Pixhawk 2.4.8 via USB
3. From popup menu click on Vehicle Setup (gears icon)
4. Click on Firmware (on sidebar)
5. Unplug Pixhawk from USB (per onscreen instructions)
6. Plug in Pixhawk to USB (per onscreen instructions)
7. Note that onscreen instructions and device information are now greyed out
8. Select ChibiOS and Multirotor
9. Click OK
10. Nothing happens. No progress, no messages. QGC is not hung as I can cancel the operation.

## System Information
When posting bug reports, include the following information

- Hardware: Intel NUC11PAHi7
- Operating System: Ubuntu 22.04.2 LTS
- QGC Version: 4.2.4
- QGC build: Stable
- Flight Controller: Pixhawk 2.4.8
- Autopilot: Currently flashed with PX4 Pro NuttX flight stack

## Detailed Description
I did find this same problem from a 2018 post (now closed) that suggested using ChibiOS instead of NuttX and that is what I am attempting here. Note that flashing the PX4 Pro NuttX flight stack works just fine as does all subsequent Vehicle Setup procedures (Airframe, Sensors, Radio, etc.).

**My Goal**
I am trying to reflash with ArduPilot so that I can perform auto tune from RC switch. PX4 Pro NuttX requires telem connection to QGC and I do not have that hardware nor a portable system to take into the field with me.

## Screenshots
[image]
[image]
[image]

booo, 2023-04-05:
Can you try latest QGC unstable and see if the problem persists there?

Maybe check your kernel log when you plugin/unplug you device and post it here. (sudo dmesg).

rtalans, 2023-04-07:
@booo 
So I just ordered GitHub for Dummies. Gotta learn how to build different branches.
I ran dmesg and was a bit shocked at the amount of info returned. I have gone over the dmesg man page but don't really know how to interpret the output. Can you point out what I should be looking for?

booo, 2023-04-07:
You don't have to build the branch yourself. Just use the image from the the CI system. E.g.:

rtalans, 2023-04-09:
@booo 

When I run the app image from the link you provided I get a new dialog shown below (the flight stack chosen is blocked by the dialog but it is the same as the above screen shot). The results are the same (fails to flash):
[image]

dmesg produced the following output. I did find it curious that two different FMU versions are listed though I really don't know how to interpret this output:

[image]

rtalans, 2023-04-11:
@booo 
I found the QGroundControl console log in Application Settings. Here is the console output from QGroundControl start->Vehicle Setup->Firmware Setup. Then I performed Steps to Reproduce 6, 8 and 9.
[ArduPilotFlashAttempt.txt](https://github.com/mavlink/qgroundcontrol/files/11198710/ArduPilotFlashAttempt.txt)

Ryanf55, 2023-04-14:
Duplicate of #10269

Same error. 
```
[E] at :0 - "QSslSocket: cannot resolve EVP_PKEY_base_id"
[E] at :0 - "QSslSocket: cannot resolve SSL_get_peer_certificate"
[E] at :0 - "QSslSocket: cannot call unresolved function SSL_get_peer_certificate"
[E] at ../../qgroundcontrol/qgroundcontrol/src/VehicleSetup/FirmwareUpgradeController.cc:680 - "PX4 releases github download failed "Error during download. Error: 6""
[E] at :0 - "QSslSocket: cannot call unresolved function SSL_get_peer_certificate"
[E] at ../../qgroundcontrol/qgroundcontrol/src/VehicleSetup/FirmwareUpgradeController.cc:769 - "ArduPilot Manifest download failed "Error during download. Error: 6""
[D] at ../../qgroundcontrol/qgroundcontrol/src/VehicleSetup/FirmwareUpgradeController.cc:249 - ""3D Robotics" "PX4 BL FMU v3.x""```

closed 2026-10-05 by github-actions[bot]