mavlink/qgroundcontrol #10651: provide a flatpak on flathub

christianrauch, 2023-04-05:
This is to let you know that I am working on a flatpak for the distribution via flathub https://github.com/flathub/flathub/pull/3968.

> - What problem does it solve?

Easier distribution and updates for users compared to AppImage.

> - What does it do?

Containerised distribution of isolated programs with their dependencies.

> - What flight stacks must it work with? (All, PX4, ArduPilot)

Not relevant.

> - Relevant vehicle types? (All, multirotor, fixed-wing, VTOL, submarine, etc.)

Not relevant.

> - Are there other systems that have this feature?

Every modern Linux distribution supports flatpak.

> - What communication/integration standards does the feature rely on (e.g. MAVLink commands etc.)

flatpak

Since the repo contains the references to all its dependencies, this is actually straightforward. A minor issue is the licence of the metadata file. Once https://github.com/mavlink/qgroundcontrol/pull/10615 has been merged the original upstream can be used and the flatpak definition can be further simplified. See the original flathub PR for instructions on how to test the current build.

There is prior work by @thopiekar (https://github.com/mavlink/qgroundcontrol/issues/10176#issuecomment-1065931745). But AFAIK, this was never submitted or published on flathub.

christianrauch, 2023-04-06:
The flatpak has been released: https://flathub.org/apps/details/org.mavlink.qgroundcontrol

booo, 2023-04-06:
I think that needs some documentation in the QGC documentation. For the users and maybe for the developers too. Could you add this?

Thanks for your contribution! Very cool.

booo, 2023-04-07:
I think somewhere here would be appropriate:

* https://github.com/mavlink/qgc-dev-guide
* https://github.com/mavlink/qgc-user-guide

christianrauch, 2023-04-07:
> I think that needs some documentation in the QGC documentation. For the users and maybe for the developers too. Could you add this?

I added user documentation at https://github.com/christianrauch/qgc-user-guide/pull/1.

I don't think there is anything special for developers. Anyone can send a PR to the flatpak repo at https://github.com/flathub/org.mavlink.qgroundcontrol to propose updates, e.g. for new releases or different build flags.

One thing that might be of interest, is a verified badge in the store (https://github.com/flathub/flathub/pull/3968#issuecomment-1497326910) to show that this is an officially supported Flatpak.

AlexKlimaj, 2023-09-21:
If we could get this building automatically in CI, that would be great.

tbo47, 2024-06-15:
Any change we can have the new version released on flatpak? @christianrauch

christianrauch, 2024-06-15:
> Any change we can have the new version released on flatpak? @christianrauch

A 4.4 release is currently blocked by https://github.com/mavlink/qgroundcontrol/issues/11611.

If released versions can be build without additional patches, the flatpak would automatically sync with the latest release.

tbo47, 2024-06-16:
oh, ok, So this doesn't need to be updated manually? https://github.com/flathub/org.mavlink.qgroundcontrol/blob/master/org.mavlink.qgroundcontrol.yml#L25

christianrauch, 2024-06-16:
No. The `x-checker-data` scans for new tags on the git repo and automatically generates (and merges) a PR.

See https://github.com/flathub/org.mavlink.qgroundcontrol/pull/11 (and specifically commit https://github.com/flathub/org.mavlink.qgroundcontrol/pull/11/commits/1b82e36459fff06ddde567875f03c923da691ce9) for a recent automatic PR for the `v4.4.0` release.

However, this only works if the project maintains the build scripts and the sources can be built without additional patches.

tbo47, 2024-06-18:
qgroundcontrol has the status of "unverified" in [flathub](https://flathub.org/apps/org.mavlink.qgroundcontrol). Which means it won't be available by default on Linux Mint.

["Unverified Flatpaks are disabled by default."](https://blog.linuxmint.com/?p=4719)

I'm not sure how much effort it would be to change that.

closed 2026-10-04 by github-actions[bot]