mavlink/qgroundcontrol #10657: Waypoints don't show while language is Chinese

jimmyxsdy, 2023-04-13:
The following template is for Issue/Bug reporting only. https://docs.qgroundcontrol.com/en/Support/Support.html#github-issues

For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources

If you are using an old stable release please upgrade first and verify your issue before reporting.
----

## Expected Behavior
I want to do a flightplan and set a series of Waypoints. As expected, Waypoints should show on map one by one after setting takeoff.

## Current Behavior
As show in the picture, Waypoints don't show.
[image]

## System Information
When posting bug reports, include the following information
- Operating System: [ Windows 10]
- QGC Version: [4.2.6(newest)]
- QGC build: [daily]

## Detailed Description
 I found that there is no that problem when QGC version is 4.0.11. What's more, when I switch language into English and reopen, everything is Ok.
[image]

booo, 2023-04-18:
Could you please check the unstable/master version of QGC? I think it is fixed there.

jimmyxsdy, 2023-04-18:
> Could you please check the unstable/master version of QGC? I think it is fixed there.
As showed in the picture，the version is v4.2.0？ But I am supposed to git clone the newest. 
[image]

RstYukine, 2023-05-09:
This question still exist in v4.2.6

booo, 2023-05-09:
Please try one of the builds from master from the CI system: https://github.com/mavlink/qgroundcontrol/actions

acgtap, 2023-05-25:
I also encountered this problem and cannot solve it

closed 2026-10-05 by github-actions[bot]