mavlink/qgroundcontrol #10663: No long press or double tap option on QGroundcontrol for Herelink

abhisheeekkk, 2023-04-18:
- What problem does it solve?

Hi all,
I wanted to setup double tap and long press on my Herelink Buttons using QGroundControl, So I researched about it and found out that QGC doesn’t have feature of double tap or long press on Herelink buttons.

Any Idea when will this feature be released. This feature is available in the SolexTx but unfortunately I am not able to connect to it.

Please, if you have any idea please let me know.

- What flight stacks must it work with? 
  Ardupilot

- Relevant vehicle types? (All, multirotor, fixed-wing, VTOL, submarine, etc.)
  All

- Are there other systems that have this feature?
  SolexTX

- What communication/integration standards does the feature rely on (e.g. MAVLink commands etc.)
   MavLink Perhaps

cbjornram, 2023-04-18:
Hi @abhisheeekkk,

I don't like to forward you to other places, but since this is related to Herelink hardware, I think your question/issue is better suited for the official Herelink-version of QGC: https://github.com/CubePilot/qgroundcontrol-herelink

Or maybe even for their forums: https://discuss.cubepilot.org/

Cheers!

abhisheeekkk, 2023-04-19:
> Hi @abhisheeekkk,
>
> I don't like to forward you to other places, but since this is related to Herelink hardware, I think your question/issue is better suited for the official Herelink-version of QGC: [...]

I Understand @cbjornram, thank you for showing me correct path.

booo, 2023-04-20:
Ok. I close this as resolved as it's not our software that needs a fix.

closed 2023-04-20 by booo