mavlink/qgroundcontrol #10666: Virtual Joystick moves when resizing QGC window (very dangerous bug)

kripper, 2023-04-22:
## Expected Behavior

Virtual Joystick should not move when just resizing the QGC window (of course).

## Current Behavior

Virtual Joystick starts moving the drone when resizing the QGC window (the virtual joystick center changes when the window is resized, so it acts as if it were triggered).

## Steps to Reproduce:

1. Enable Virtual Sticks and Auto-Center Throttle
2. Move the drone once (it will auto-center after that)
3. Resize the QGC Window
4. Look at the joystick widget. It won't be centered anymore.

## System Information
When posting bug reports, include the following information
- Operating System: Windows 10
- QGC Version: v4.2.4
- QGC build: stable
- Flight Controller: Any
- Autopilot (with version): Any

## Detailed Description

After resizing the window, the center changes, so the Virtual Stick acts as if it was triggered.

DieBorr, 2023-05-26:
I have tested it on Ubuntu 20.04.5 and yes, I have noticed this behavior

DieBorr, 2023-05-26:
I think I have resolved it, I will push the commit that resolves this issue
 

https://github.com/mavlink/qgroundcontrol/assets/47715559/ff50f85e-d10d-45e0-86bd-f2cbfc2444d1

kripper, 2023-05-26:
Yes, seems to be fixed.

closed 2024-05-31 by HTRamsey