mavlink/qgroundcontrol #10681: Video streaming is not working on steam deck, with plugins installed

maximilian101, 2023-05-08:
The following template is for Issue/Bug reporting only. https://docs.qgroundcontrol.com/en/Support/Support.html#github-issues

For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources

If you are using an old stable release please upgrade first and verify your issue before reporting.
----

## Expected Behavior
Video output streamed
## Current Behavior
No video
## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. installed gstreamer gst-plugins-base good bad ugly libav after disabling the readonly and populated the keys
2. connect the regular parrot anafi with wifi
3. no videos streaming, but drone can be controlled with the joysticks on steam deck

## System Information
When posting bug reports, include the following information
- Operating System: steamos 3.5
- QGC Version: 4.0.0 to 4.2.6
- QGC build: appimage provided
- Flight Controller: unsure, steam deck? not connected to skycontroller
- Autopilot (with version): unsure, PX4 pro?

## Detailed Description
Provide further details about your issue/bug.

## Log Files and Screenshots
- [QGC Console Logs](https://docs.qgroundcontrol.com/en/Support/Support.html#reporting-bugs)
- Autopilot logs when available (post a link).
- Screenshots of QGC to help identify the current issue/bug behavior.

Debug 1
in console
`gst-launch-1.0 --gst-debug 1 udpsrc address=localhost port=5600 ! application/x-rtp, media=video, clock-rate=90000, encoding-name=H264 ! parsebin ! tee name=t ! queue ! valve ! decodebin3 ! autovideosink t. ! queue ! valve ! matroskamux ! filesink location=teste05.mkv
`
```
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
ERROR: from element /GstPipeline:pipeline0/GstUDPSrc:udpsrc0: Internal data stream error.
Additional debug info:
../gstreamer/subprojects/gstreamer/libs/gst/base/gstbasesrc.c(3127): gst_base_src_loop (): /GstPipeline:pipeline0/GstUDPSrc:udpsrc0:
streaming stopped, reason not-linked (-1)
Execution ended after 0:00:07.772059603
Setting pipeline to NULL ...
Freeing pipeline ...
```

FixMe from QGC
```
GStreamerAPILog: <source> error: Error (500): Internal Server Error
VideoReceiverLog: GStreamer error: Could not write to resource.
GStreamerAPILog: <source> setup failed
GStreamerAPILog: <source> can't setup streaming from sdp
GStreamerAPILog: <source> failed to open stream
GStreamerAPILog: <source> error: Error (500): Internal Server Error
VideoReceiverLog: GStreamer error: Could not write to resource.
GStreamerAPILog: <source> setup failed
GStreamerAPILog: <source> can't setup streaming from sdp
GStreamerAPILog: <source> failed to open stream

```
The CAM parameters seems missing as well.

gailingmic, 2023-06-09:
the issue is that the flatpak doesnt bundle the gstreamer plugins. however i had no success compiling it directly on the deck aswell... (and i wouldnt reccomend anyone doing it)

JustTSK, 2023-06-09:
> the issue is that the flatpak doesnt bundle the gstreamer plugins. however i had no success compiling it directly on the deck aswell... (and i wouldnt reccomend anyone doing it because you have to disable quite alot of nice steam features to do so)

I was able to compile it just fine after installing all necessary dependencies but steamdeck clears the system packages every time there's an update so I had to redo this every time.
What steam features are you talking about when you say you have to disable them?

maximilian101, 2023-06-10:
Still didn’t find a solution to run the appimage on steam deck. I wonder if it is possible to wrap the source and required dependencies into a SteamOS specific appimage, but I don’t have the knowledge to work on that yet. The immutable SteamOS is really hard to work with, and it wipes everything clean after updates. Btw, one workaround is to install the exe through wine, most of the functions including the built-in controller and video streaming works fine, but the app can’t be launched from game mode through proton, and the app crashed once in a test flight.

fraviofii, 2023-07-20:
I've built QGC on a Debian, generated the appImage to a SteamDeck.

I was able to unlock the filesystem and installed gstreamer and its components manually as well. QGC works correctly on SteamDeck.

But it is not working due on my end due incompatibility with the streaming server. It is a pipeline issue, because I was able to get the streaming from the command line.

closed 2024-05-31 by HTRamsey