mavlink/qgroundcontrol #10682: mavlink forwarding is one-way

tridge, 2023-05-09:
With current 4.2.6 release if you forward mavlink over UDP then it gets a one-way link. It does send out mavlink packets from the link to the flight controller, but does not fwd packets arriving over the link to the flight controller
It needs to be bi-directional for support.ardupilot.org to work
tested on windows version 4.2.6

Davidsastresas, 2023-05-09:
Hello Andrew, thanks for reporting. I tried to reproduce it as follows:

- QGC: forwarding mavlink to localhost, port 15555
- AP SITL connected to QGC using the usual 5760 port
- Instance of mavproxy connecting to UDP 127.0.0.1 port 15555

With the above setup I am able to send commands from the standalone mavproxy instance, and they work. This should simulate what you meant right? If not, how can I reproduce it locally?

I am on Ubuntu 20.04.

I attach a short video of my test, using QGC 4.2.6 and Ardupilot master:

https://drive.google.com/file/d/1x2X_X3myw4q4nYotdkokDq-wmsWgkrCs/view?usp=sharing

Should this test reproduce what you are describing? If so I will repeat it from windows to see if I can reproduce it there.

Thanks!

tridge, 2023-05-10:
it fails for me on both linux and windows. Can you try using support.ardupilot.org, see if you can reproduce?

SebastianViollaz, 2024-01-24:
@Davidsastresas @tridge 
I have the same problem but only with non simulated UAVs.

 I have installed the SITL of ardupilot and run the Arducopter sim. I that case i can connect to the mavforawding link and i'm able to send and extract data from the fligthcontroller and all works fine.
 The problem is when i connect to my real UAV. In this case the link is only for extracting data from the fligth controller and no to send data to it. 
 
 I already tried to:
 -Update QGC to LTS
 -Update Ardupilot Firmware
 
 Im runing my app that try to communicate with QGC in linux Ubuntu 22.04.
 
 
 Any ideas if it a configuration problem? QGC problem?

Davidsastresas, 2024-03-25:
@SebastianViollaz, could be that the sys id of the secondary GCS could not be what your sysid_mygcs parameter was? with that parameter to enforce GCS sysid enabled?

In current master, and Also in latest stable, this should not be a problem. Let me know if you find out more so we can close the issue. Thanks!

ymoumen, 2024-05-27:
I have the same problem. All works  well except that the flight controller doesn't receive commands. The link seems to be one way.
Configurations tested:
- SITL with UDP 14550 on ArduCopter 4.4.4, QGroundControl 4.3.0 as relay and Mission planner latest
- SITL with UDP 14550 on ArduCopter 4.4.4, QGroundControl 4.3.0 as relay and MAVProxy
- USB on CubeOrange+ with ArduCopter 4.4.4, QGroundControl 4.3.0 as relay and MAVProxy/Mission planner
- SIYI MK15 connected to CubeOrange+ with ArduCopter 4.4.4, QGroundControl 4.3.0 as relay and MAVProxy/Mission planner
None works with the same issue of Autopilot not receiving commands (one way).
I tried both MAVLink forward and RemoteSupport features with the same results.
I tried on Ubuntu 22LTS and Windows 10 using all in localhost, on local network and Zerotier.
I tried to disable UDP autoconnect in QGroundControl.
I tried many combinations of system ID for all GCS and flight controllers.
I'm tempted to say this is a QGC forwarding or flight controller code logic issue.
This feature would be very useful for us to support clients using QGC on SIYI MK15 radio controllers. I need to used FTP over MAVLink to update scripts. The only way to do it now is to install Mission Planner on Android but it's messy.
I'm going through QGC code to see if I can debug this. If I find some solution I'll share it from my side.

michaellmonaghan, 2024-06-06:
I'm seeing the same behavior on MacOS.

> I have the same problem. All works well except that the flight controller doesn't receive commands. The link seems to be one way. Configurations tested:
>
> * SITL with UDP 14550 on ArduCopter 4.4.4, [...]

doquangquan93, 2025-03-11:
Hello everyone. Is there a solution to this problem?

DonLakeFlyer, 2025-03-11:
Use a daily build like the comments say?

michaellmonaghan, 2025-03-23:
I take it changes from daily made it into 4.4. I'll give this a try next time I have the chance.