mavlink/qgroundcontrol #10702: Please add a decent DEM source (the current ends at 50°)

AndKe, 2023-05-28:
The Ardupilot project has decent world coverage at  https://terrain.ardupilot.org/ 

QGC's current Airmap source is limited to 50°N to 50°S - this may be sufficient for flat-earthers, but otherwise, it's simply inadequate and sad.

The elevation graph is a neat safety feature that does not work for huge areas due to a very limited data source.

Davidsastresas, 2023-06-01:
It seems Airmap will shut down the business, so in a few weeks we won't have the service available anyway... We should do something about it!

AndKe, 2023-06-01:
@Davidsastresas  I wonder about the precision/grid size of the source data, the preview mapbox provides is very impressive, but maybe just very interpolated.
But it is a huge improvement over "nothing".
They also reveal that all the tiles are 0kb !
[image]

This means we can precache the whole world at maximum zoom, and use no space to do so!   Incredible!  :laughing:

TorbjornHouge, 2023-08-02:
It would be useful to allow for alternative data sources for elevation. Many countries may have their own terrain models that can be quite good (Norway has an excellent one as an example), and many professionals tend to have their preference with which one they would like to utilize if possible. If it is possible to allow for the most common standards and then let the user input their links/credentials/tokens for using the specific ones, that would be fantastic.

dayjaby, 2024-02-13:
Any updates here? Airmap not only shut down their business, but deleted their organization/repositories on github.

Davidsastresas, 2024-02-14:
> Thanks @Davidsastresas . I also found the PR #10740 , which appearently also fixes the issue. Or what's the difference between the Airbus and Auterion terrain servers?

That is precisely the PR I was talking about. Auterion fix involves those Airbus maps. I will close the issue then, thanks!

closed 2024-02-14 by Davidsastresas