mavlink/qgroundcontrol #10707: mode change dangerous for some modes

tridge, 2023-06-02:
the current mode change drop down is potentially dangerous, with users sometimes choosing modes like ACRO which will result in an instant crash for users not prepared for them.
I don't think adding confirmation for all modes is the right answer, as sometimes you do need fast mode switching.
Maybe this?
 - have a way to disable some modes completely so they don't appear in the list
 - be able to mark some modes as requiring confirmation (eg. MANUAL mode or QSTABILIZE on a VTOL plane)
thanks to alex from ARACE for this image:
[image]

tridge, 2023-06-02:
another option is to have a list of "favourite" flight modes. Favourites would be in that drop down if favourites is non-empty (otherwise would have full list). Non-favourite modes would be via a separate menu with confirmation

DonLakeFlyer, 2023-07-05:
FYI: I have favorite flight modes already implemented in a pull. I was waiting for a new stable to go out before I push it through. It basically sets some flight modes as advanced (including Acro for xeample). By default these advanced flight modes are not included in the dropdown by default. You can then edit the dropdown to add/remove what you want.

It's all sitting in here: https://github.com/mavlink/qgroundcontrol/pull/10618

DonLakeFlyer, 2024-08-20:
I made all mode changes from the toolbar require confirmation. We'll see if that causes problems or not.

PR #11793 (2024-08-21): Flight mode confirmation
Fix #10707
Fix #11547
closed 2024-08-21 by DonLakeFlyer