mavlink/qgroundcontrol #10727: Running ForceVideoDecoderDirectX3D on Linux

fraviofii, 2023-07-09:
I have a RTSP camera which works on QGC on Windows only if the option `ForceVideoDecoderDirectX3D`. If I use other options, I cannot receive camera images.

And I need to have QGC on Linux and it is not working. I tried all the available options, without success.

I saw on the source code that the option `ForceVideoDecoderDirectX3D` is not available for Linux. Is there any way to enable this feature?

patrickelectric, 2023-07-09:
DirectX is a Windows API, does not exist under Linux. Make sure to check the log under Linux to see if you have the video drivers and the required hardware to use the video options available.

fraviofii, 2023-07-10:
I can see that this flag disables some decoding options in gstreamer.

It it possible to have such decoding options on Linux and I can used them?

patrickelectric, 2023-07-10:
The options only have effect under windows, they are windows specific code.

As said previously, be sure to have your system video drivers installed on linux and use the options available on linux.

fraviofii, 2023-07-10:
How do I know if I have the correct video drivers?

when I run using gstreamer directly, it works using the following command:

`gst-launch-1.0 rtspsrc location=rtsp://drasp:34554 latency=100 ! queue ! rtph264depay ! h264parse ! avdec_h264 ! autovideosink`, it works.

But when I run: `gst-launch-1.0 rtspsrc location=rtsp://drasp:34554 latency=100 ! queue ! valve ! rtph264depay ! h264parse ! autovideosink`, it does not work.

Can we conclude anything with this information?

fraviofii, 2023-07-20:
@patrickelectric , do you know how to customize the gstreamer pipeline in QGC?

I tried to remove the valve from the pipeline, without success.

I was thinking about the possibility to replace the current pipeline with one used by `gst-play-1.0`, for a specific option for my project, but I have no idea about how to do that.

Can you help me on this challenge?

closed 2024-05-31 by HTRamsey