mavlink/qgroundcontrol #10730: qgroundcontrol 4.2.7 running on CLion // CMake 's bug

whp2000, 2023-07-11:
Maybe because the Airmap was deleted,but I don't know where is the root reason caused the next error: 

====================[ Build | QGroundControl | Debug-MinGW-Qt ]=================
"D:\Clion\CLion EAP\bin\cmake\win\x64\bin\cmake.exe" --build E:\qgroundcontrol\cmake-build-debug-mingw-qt --target QGroundControl -j 10
[1/1] Linking CXX executable QGroundControl.exe
FAILED: QGroundControl.exe 
cmd.exe /C "cd . && D:\Qt\Tools\mingw810_64\bin\c++.exe -g  @CMakeFiles\QGroundControl.rsp -o QGroundControl.exe -Wl,--out-implib,libQGroundControl.dll.a -Wl,--major-image-version,0,--minor-image-version,0  && cmd.exe /C "cd /D E:\qgroundcontrol\cmake-build-debug-mingw-qt && "D:\Clion\CLion EAP\bin\cmake\win\x64\bin\cmake.exe" -E env PATH="D:/Qt/5.15.2/mingw81_64/bin" D:/Qt/5.15.2/mingw81_64/bin/windeployqt.exe --verbose 0 --no-compiler-runtime --no-angle --no-opengl-sw "E:/qgroundcontrol/cmake-build-debug-mingw-qt/QGroundControl.exe" && cd /D E:\qgroundcontrol\cmake-build-debug-mingw-qt && "D:\Clion\CLion EAP\bin\cmake\win\x64\bin\cmake.exe" -E copy_if_different E:/qgroundcontrol/libs/sdl2/msvc/lib/x64/SDL2.dll E:/qgroundcontrol/cmake-build-debug-mingw-qt""
D:/Qt/Tools/mingw810_64/bin/../lib/gcc/x86_64-w64-mingw32/8.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: cannot find -lAirmap
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

zdanek, 2023-07-11:
`cannot find -lAirmap`

whp2000, 2023-07-11:
Yeah,I know that cannot find lAirmap,but I don't know what cause the problem,today's version has removed the Airmap module,why this error still happened, I really want to know

whp2000, 2023-07-11:
> `cannot find -lAirmap`
> 
> 

Yeah,I know that cannot find lAirmap,but I don't know what cause the problem,today's version has removed the Airmap module,why this error still happened, I really want to know

HTRamsey, 2024-05-31:
I believe some remnant airmap stuff has been removed and this shouldn't be an issue anymore. Please reopen if it is.

closed 2024-05-31 by HTRamsey