mavlink/qgroundcontrol #10743: Guided Altitude Slider Linear Slider with Snapping

JMare, 2023-07-21:
This is a continuation of a short discussion on Discord, and is my concept for an improvement to the guided altitude slider.
This is something I have modified in a custom build already, but I am making this issue to discuss whether it would benifit others and would be appropriate for integrating into master.

## Background
The current exponential altitude slider is designed to "allows for fine adjustment around the current altitude, but also allows the upper/lower end of the range to hit a full range adjustment" see #5150 for background.

## The issues with the current design
My suggestions are based on my experience of deploying a fleet of multirotors flown exclusively with guided mode through qgroundcontrol (On a 10" tablet). The drones are flown between 0 and 120m. We have found the following issues with the current behavior:
- The exponential slider makes it difficult to precisely make large altitude changes, as the resolution decreases away from center. While it is very easy to "go up 1.5m" it is hard to precisely "go to 80m" which at least in our case is a more common scenario.
- The non linearity makes for a less predictable experience and requires the user to concentrate quite hard to dial in specific altitudes.
- The change altitude slider behaves differently from the takeoff slider

## Solution Mockup
In my custom build, I have written the following: The slider snaps to the tickmarks and the red section shows the minimum guided height, below which you cant drag the slider. I find it works quite well. This is similar to what was suggested in #5795.
[image]

## When would this not work?
While I appreciate that there may be scenarios in which the exponential slider is a better option for this, I believe they are the minority of use cases. I believe the solution I am proposing would not work when:
- High altitude precision is required - eg the ability to go to 43m exactly, rather than being content with coarser altitude steps
- A very large altitude range is in use, beyond the normal 0-120m
- The vehicle needs to fly below the home point

## My suggestion for integration
- Under the guided command settings section, a drop down to select linear vs exponential slider (default whatever is deemed appropriate for most users)
- a tickbox to enable snapping on the linear slider if desired (default enabled)
- a setting for slider snap resolution, ie a setting of 10 will snap to the nearest of 10 values. We can setup the defaults for this setting to be low for mobile builds, and high for desktop builds. eg a value of 16 is what is used in my example above and is practical on desktop/laptop, but on a phone a value of 8 eg 50ft resolution would probably be more usable. (we could also limit the maximum allowed value depending on platform)

DieBorr, 2023-09-06:
Hi, I have red your discussion and i have came with the next idea: 
Apart from giving the user the chance for deciding between linear and exponential slider I think that it would be also useful to give him the chance to introduce the exact altitude value in a popup when he does an specific action.
As you said in small screens is difficult to select an specific altitude, linear slider would make this easier but probably not easy enough to select it without trying a couple of times, at leas in my opinion.
In this days I will try the Pop Up Idea, any suggestions of the action to activate the Pop Up?

HTRamsey, 2024-05-31:
I really like this

JMare, 2024-06-10:
I would be happy to get a pull request ready if the change would be welcomed.
I have the bare bones of the code in a branch somewhere - it would need polishing to be ready for a PR.

I also definitely agree with DieBorr's comment that entering an exact altitude would be great, I also tossed around the ideas of having +- buttons to go up or a down a fixed amount ie a button to go up 50ft or down 50ft. Most of the time I am flying around in guided mode, ease of use is more important than precision on the altitude slider.

From memory, a lot of the complexity with the snapping comes from trying to make the ticks fall on nice even numbers, for different max altitudes and with different units. 
I think I ended up with the easiest way being to configure it based on divisions rather than actual values. EG, if the users max altitude was 400ft, they could choose 8 or 16 divisions, but if the max altitude was 350ft, the user would need to choose 7 divisions to get it to land on nice numbers.

I think there was also a change to the vehicle API that would be best to do to support it, to be able to send absolute guided commands rather then relative ones - I think currently altitude changes get sent as relative commands which would be less suitable for this type of control.

Happy to discuss the desired functionality and then implement it. 😄

DonLakeFlyer, 2024-06-11:
My concern here is that this change brings back the problem which was one of the main drivers for the non-linear slider: "Using a large linear range slider on a small screens creates large jumps in altitude for small movement of the slider". The solution needs to be able to scale down to a Herelink sized display.

Ages ago the altitude slider has a text entry field for exact altitude setting. It was removed at some point for some reason which I don't remember. Would need to go back through pulls to see if there are details as to why that was a problem.

I think it would be interesting to first have a discussion around the user scenarios that are trying to be solved here. As opposed to jumping directly to a ui solution. Some notes around that discussion:
* "Most of the time I am flying around in guided mode, ease of use is more important than precision on the altitude slider."
* "While it is very easy to "go up 1.5m" it is hard to precisely "go to 80m" which at least in our case is a more common scenario."
* I also remember from the original discussion around change altitude was the ability to quickly duck down to a lower altitude, say when a real aircraft comes into the drones airspace 
* What else...

AndKe, 2024-06-11:
I like the mockup, and I have a suggestion to mitigate all three "issues":

"high precision is required / 
A very large altitude range is in use, beyond the normal 0-120m /
The vehicle needs to fly below the home point"

A "very high" precision is usually pre-programmed, so we can limit the resolution to 10m / 25ft  (round numbers)
Let's say the displayed scale is initially -20 ... 120m    
if user selects 30m , ... same scale.

if user selects a value in the upper 80%  , (let's say 100m)- then expand the scale to :
(selection/2) to (2x selection)  == (50...200m) ,  and so on.
this way, if you want to command 300mREL , then  you slide the slider to 120m,  - the tool would scale to 60...240m  , user selects again 240, and it would scale to  120...420m   

opposite for selecting something in the lower 20% of the tool.
This method would keep commanded altitude near middle.

We could just copy the logic and make it to "zoom in" if the changes were near the center of the tool: 
changing altitude within less than 30% of the graph height, could shrink the tool's  min/max.

Likewise, a smaller change, let's say between 60-80% of the graph ()

DonLakeFlyer, 2024-06-12:
Yeah, all makes sense. One caveat I would add is that there should be some lockout/warning on setting an altitude below the home point. Worried about users screwing that up when they really don't have more space to fly below home alt.

AndKe, 2024-06-12:
@DonLakeFlyer  maybe one of yours "slide to confirm" boxes should be spawned on the first attempt to do so?
 "Drag to confirm negative altitude target"

JMare, 2024-06-19:
I've had a good think about this

I think we should perhaps consider two scenarios

- Normal flight profile - +2m to +122m guided altitude range (qgc defaults)
- Extended flight profile - Lets say -100m to +500m as what an advanced user might have

Both of these flight profiles should be well supported on both small and large displays

- "My concern here is that this change brings back the problem which was one of the main drivers for the non-linear slider: "Using a large linear range slider on a small screens creates large jumps in altitude for small movement of the slider". The solution needs to be able to scale down to a Herelink sized display."
- I think that its reasonably easy to handle a herelink sized display provided we use the normal flight envelope with a reasonable level of precision eg 20 or 30m snaps perhaps.

--- 

@AndKe, I like your idea and I think it could be a good solution especially for users that normally fly in a normal range but occasionally want to go higher or lower. It would also probably scale well to any screen size if the default displayed altitude window is sized to fit the screen dynamically... EG herelink might by default have a 60m window while desktop 120m. I want to get my head around all the edge cases though especially around guided min/max settings

**Normal flight profile:**
- During takeoff, what would the extents of the slider be? You mentioned -20 to 120 but presumably the guided altitude min max should be respected. Either the slider range could be reduced or the bottom extent could be blocked out like in my mockup in red.
- Once in flight, what should the slider adapt to when:
- - The aircraft is flown to 50m: Zoom in?
- - The aircraft is flown to 10m: Expand downwards or respect guided min/max
- - The aircraft is flown to 110m: Expand upwards or respect guided min/max
- Usability Questions
- - For someone flying in this flight profile - I would think 10m is an appropriate altitude step
- - Would a zoom in around 50m be desired while they are flying? This would be trading range for precision.

**Extended flight profile**
- During takeoff: Presumably we need to prevent takeoff commands with negative altitude, so 0-120 or -20-120 with a blocked out region? (Again possibly device size dependent)
- During flight:
- - Aircraft is flown to 50m: Zoom in?
- - Aircraft is flown to 10m: Expand downwards, eg to -100 to +50m?
- - Aircraft is flown to 110m: Expand upwards, eg to +50 to +200 like your example?
- Usability questions
- - For someone operating in such a wide altitude range, up to 500m, would 10 meter altitude steps be needed?
- - someone would potentially need to climb to 300m with three movements as the slider range moves up.

 --- 

I think the risk of trying to create single solution that caters for the extended flight profile is that the result will be less predictable, repeatable and intuitive for the normal flight profile. I personally believe (without any evidence to back it up) that the vast majority of users fly in the normal flight profile. That said, I'm open to other solutions than what I first proposed and it would be great if we can find one that works well for all cases.

From memory of when I wrote my version of this, there are also some technical limitations of the qt slider class and how it handles snapping - being that it only accepts a "divisions" value and will evenly divide the slider range. This means that a scale of 2-122m with snaps at 10m,20m.... would be impossible without doing a custom slider implementation.

DonLakeFlyer, 2024-06-20:
> there are also some technical limitations of the qt slider class and how it handles snapping

This is going to require a completely custom control in order to do it will.

DonLakeFlyer, 2024-06-20:
I'll take a stab at a prototype for this. I'm fried from fighting with cmake build systems stuff. Need to do something fun for a while.

DonLakeFlyer, 2024-06-24:
Here is a gif of what I'm working on. Basically the slider has multiple ranges it can work through (amount of value change without major ticks). You drag it up down with mouse/finger to change the value. And then you can adjust the range using either scroll wheel/pinch zoom to bring the scale of values up/down. This way you get fine grained and card grained adjustment by zooming the slider in/out.

Still a mile of work to do on it. But the functional concept is working...

[image]

AndKe, 2024-06-25:
@DonLakeFlyer 
"scroll wheel/pinch zoom to bring the scale of values up/down"
Great idea, looks good, maybe a small text like "zoomable" can tell users about this feature?
Also:  when dragging, it could be nice to have a "throw" logic; an option to "make it scroll"  until clicked.  
Basically: if click/touch is lifted while it is being scrolled at some velocity, continue to scroll and brake for a few seconds.

DonLakeFlyer, 2024-06-25:
> maybe a small text like "zoomable" can tell users about this feature?

Yeah, that's been in the back of my head to do.

> it could be nice to have a "throw" logic;

Yeah, I was going to add flick support

DonLakeFlyer, 2024-06-26:
And then the whole flick thing might make the zooming stuff not needed. If you can flick for large distance scrolls, but still have precision positioning when it stops then it might not be needed. I'll try it.

DonLakeFlyer, 2024-06-28:
I have a full implementation of this in the pull now:
* I set the flickable deceleration value to a high value. This allows you to quickly flick through large value changes. With this the whole need for multiple scales is not needed I believe.
* This does not add new support for things like change alt below home position (I'll get to that later)
* If the units are meters it allows for changing values to one decimal place. If feet, it allows for integral changes only.

I have tested this against PX4 SITL. I could use help testing against ArduPilot SITL. Also @Davidsastresas I could use help testing the speed change support you added.

JMare, 2024-07-02:
@DonLakeFlyer I really like this.

I just gave it a test on ardupilot SITL, seems to work well and i think it will be a big improvement

Minor things: I somewhat think that the takeoff setpoint should be in the middle like the change alt one for consistency.
and i'm not sure there would be a way to detect if the touch was from a mouse or a touchscreen?
The inertia makes sense for touch but a more rapid stop might be good with a mouse.

I can do some more testing with Ardupilot - is the mavlink:GuidedValueSlider the most up to date code?
On this branch i was able to fly below the guided minimum setting.

AndKe, 2024-07-02:
Good work, I am on vacation till August, and unable to help testing.

JMare, 2024-07-02:
> > is the mavlink:GuidedValueSlider the most up to date code?
> > On this branch i was able to fly below the guided minimum setting.
> 
> That should not have been allowed. You did a change alt below minimum? I'll check it.
> 
> > I somewhat think that the takeoff setpoint should be in the middle like the change alt one for consistency.
> 
> I was thinking that since the starting point for takeoff slider is a minus takeoff alt all the values below are invalid. And that would in turn provider more resolution to adjust the slider in the valid space. But it's probably fine to have it be in the middle like everything else. I'll change that.

I think a greying out of the scale outside the guided alt min/max range could be a good way to see the extents.

Yes, I had had default guided settings (2m - 122m) and i was able to command change alt to -1.1 and hit the ground. Arducopter Sitl on linux. Just checked it again to make sure

Davidsastresas, 2024-07-04:
I fixed some issues with the slider for Ardupilot, both for altitude and speed change. More testing would be welcomed! specially for PX4. Thanks!

PR #11641 (2024-07-08): Guided value slider
Fixes #10743

See Issue for details
closed 2024-07-08 by DonLakeFlyer