mavlink/qgroundcontrol #10755: QGC terminates, while trying "Use Multi Channel Mode Selection

jstrebel, 2023-08-03:
## Current Behavior
QGC terminates without any log entry

## Steps to Reproduce:
Click at the Button

## System Information
- Operating System: tried with WIN, Android,OSX
- QGC Version: [e.g. 4.2.8]

- Flight Controller: [e.g., Pixhawk I, Pixhawk Mini, Pixhawk 2, etc.]

## Log Files and Screenshots
No crash report is generated
[image]

DonLakeFlyer, 2023-08-04:
Is this still supported by the firmware? It think it may have been removed.

DonLakeFlyer, 2023-08-04:
This crashes on stable because this feature no longer exists on the firmware side. Daily builds will work fine, because it just not there!

closed 2023-08-04 by DonLakeFlyer