mavlink/qgroundcontrol #10774: Problem running RTSP on Windows11, and trick to solve it

TianmingLi, 2023-08-15:
The following template is for Issue/Bug reporting only. https://docs.qgroundcontrol.com/en/Support/Support.html#github-issues

For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources

If you are using an old stable release please upgrade first and verify your issue before reporting.
----

## Expected Behavior
RTSP Video should appear on the QGroundControl screen.

## Current Behavior
RTSP Video not showing up for my specific RTSP Server.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. SET RTSP address
2. Open QGC
3. Video Not appear
4. Close QGC
5. Open VLC, Open RTSP stream, Pause stream
6. Close VLC
7. Reopen QGC

https://github.com/mavlink/qgroundcontrol/assets/5005440/5ec5c9e3-5685-48a7-b46e-4a179d61756e

[video_1080p.zip](https://github.com/mavlink/qgroundcontrol/files/12340962/video_1080p.zip)

## System Information
When posting bug reports, include the following information
- Operating System: Windows11
- QGC Version: 4.2.8
- QGC build: stable
- Flight Controller: Pixhawk
- Autopilot (with version): PX4 1.13.3

## Detailed Description
Provide further details about your issue/bug.

## Log Files and Screenshots
- [QGC Console Logs](https://docs.qgroundcontrol.com/en/Support/Support.html#reporting-bugs)
- Autopilot logs when available (post a link).
- Screenshots of QGC to help identify the current issue/bug behavior.

[video_1080p.zip](https://github.com/mavlink/qgroundcontrol/files/12340962/video_1080p.zip)

TianmingLi, 2023-08-15:
I tried two laptop, they all had same problems , and they all can be solve by this method. 🙃

Luther999, 2024-11-19:
Hi bro, have you solved this problem? I see a lot of people with the same problem, but don't see a solution.

closed 2025-05-03 by HTRamsey