mavlink/qgroundcontrol #10777: gstreamer works well in QGC but cannot use gstreamer outside

jashshah999, 2023-08-16:
## Expected Behavior
Both, QGC and gstreamer should give the same output. However, QGC gives a good output, gstreamer does not give a good output, laggy and choppy (videos below). 

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Using QGC v4.2.8. 
2. Give H.264 UDP stream with correct port on QGC, get good stream
3. Run `gst-launch-1.0 udpsrc port=5600 ! application/x-rtp ! rtpjitterbuffer ! rtph264depay ! avdec_h264 ! videoconvert ! autovideosink` and get choppy output

## System Information
When posting bug reports, include the following information
- Operating System: Ubuntu 20.04
- QGC Version: 4.2.8

## Log Files and Screenshots

[Screencast from 08-15-2023 11:03:07 PM.webm](https://github.com/mavlink/qgroundcontrol/assets/49280550/d9f92fe5-e039-4ae3-9b18-3ac7b2f8e750)
[Screencast from 08-15-2023 11:05:14 PM.webm](https://github.com/mavlink/qgroundcontrol/assets/49280550/042a6ec0-2f63-42be-83ce-f21d5ab61490)

dakejahl, 2023-08-18:
remove the jitterbuffer and try adding a queue between the depay and decode. If that doesn't work do some googling to try and tweak the pipeline
```
gst-launch-1.0 udpsrc port=5600 ! application/x-rtp ! rtph264depay ! queue ! avdec_h264 ! videoconvert ! autovideosink
```

jashshah999, 2023-08-18:
Hey, I did try this and lot more combinations but it turns out none of them work. However, suprisingly, everything works well on QGC. It is very weird.

closed 2024-04-05 by HTRamsey

alyshowoliveira12, 2024-11-25:
Same over here. Did you get to solve this?