mavlink/qgroundcontrol #10779: QGC COMM LOST WITH RED BAND & Disconnect display

sibujacob, 2023-08-16:
## Expected Behavior
While rebooting fc after parameter changes under param ---》 Tools, QGC is expected to show a GREEN BAND with manual connect.

## Current Behavior
After teboot, QGC displays a RED band with communication lost & Disconnect message althpugh yhe drone is pretty much in pre arm state , ready to arm . Thos may become a safety issue for pilots

## Steps to Reproduce:
Following video indicates the actual error
1.https://youtu.be/L6XJY4Rxcgw

## System Information
When posting bug reports, include the following information
- Operating System: Android 
- QGC Version: [ Custom QGC HERELINK VERSION ]
- QGC build: [daily, stable, ..]
- Flight Controller: [e.g., Pixhawk orange CUBE.]
- Autopilot (with version): [PX4 1.12.3]

## Detailed Description

## Log Files and Screenshots

DonLakeFlyer, 2024-02-26:
Not quite sure if this is fixable. The QGC states doesn't really relate to what the sticks may or may not do since the sticks are not managed by QGC.

sibujacob, 2024-02-27:
OK...Anyway the issue was not seen with subsequent releases of the QGC v42.8

DonLakeFlyer, 2024-02-28:
Closing then

closed 2024-02-28 by DonLakeFlyer

sibujacob, 2024-02-28:
Yes....thank you