mavlink/qgroundcontrol #10800: Inquiry Regarding QGCVideoReceiverAPP Code Issues

HEIseYOUmolc, 2023-09-04:
### I want to find the author of this code, but I am unable to send an email to their address. Therefore, I would like to post here in the hope that others can help me mention or contact them.
### 
 @file
 *   @brief QGC Video Receiver
 *   @author Gus Grubba ### <gus@auterion.com>

### Here is my contact information:
Hello, I came across your code on GitHub, and it seems like it's been a while.

I'm not sure if you're still maintaining this project. 

First of all, I want to express my gratitude for your contributions. 

Secondly, I have some code-related questions. I tried the QML approach you provided, but with QT 6.5, it seems that many of your previous designs have been deprecated. 

Attached is an image of a simple QTUI interface I created. I encountered significant issues when trying to run it, which might be minor for you. 

I'd like to invite you to help me address them.

If you are willing, you can reply to me at this email address or my personal email: [1615510548@qq.com](mailto:1615510548@qq.com).

HEIseYOUmolc, 2023-09-04:
### This is the attachment mentioned in the document.
[image]

HEIseYOUmolc, 2023-09-04:
@andrewvoznytsa

HEIseYOUmolc, 2023-09-04:
`void VideoReceiverApp::URLChanged(const QString& URL)
{
    _url=URL;
    ui->info->setText("更换URL:"+ _url);
}`
The assignment I make to _url causes the program to crash. Where should I look for a solution, and if there is anything else I need to provide, I will do so promptly.

andrewvoznytsa, 2023-09-04:
I think this app is not maintained anymore (at least not by me). Most likely you'll need to debug it on your own.

HEIseYOUmolc, 2023-09-04:
> I think this app is not maintained anymore (at least not by me). Most likely you'll need to debug it on your own.

I'm glad you responded. If possible, could you provide me with some references you used when working on your project? :) If you still have them, perhaps I'll update some of my questions, and your responses can help me come up with new ideas. Either way, thank you for your contributions.

HEIseYOUmolc, 2023-09-04:
`void VideoReceiverApp::run()
{
    if((_videoSink = GStreamer::createVideoSink(nullptr, ui->widget)) == nullptr) {
        qDebug() << "create VideoSink failed";
        return;
    }
    _receiver->startDecoding(_videoSink);
}`What puzzles me is the triggering of this piece of code. First, I'd like to mention that I'm relatively new to Qt, having just started working with it three months ago. My understanding of QThread is quite limited, but I've been trying to adapt the widget from the original QML to point to the widget in my UI by reading the code you provided.

DonLakeFlyer, 2023-09-04:
Gus hasn't worked on QGC for years. He also is no longer available for questions. Andrew isn't around any more either as far as I know. There are a number of other folks who have taken over the video code. Best place for questions on that is the QGC discord channel.

HEIseYOUmolc, 2023-09-05:
> Gus hasn't worked on QGC for years. He also is no longer available for questions. Andrew isn't around any more either as far as I know. There are a number of other folks who have taken over the video code. Best place for questions on that is the QGC discord channel.

Thank you for your response. I will go to the Discord channel to further explore these issues. Could you please help me locate the channel I should join? Currently, I have selected the DRONECODE FOUNDATION #dev-team channel.

HEIseYOUmolc, 2023-09-07:
`GstElement* videoSink = GST_ELEMENT(sink);`The latest bug, this code null pointer to GstElement is not a regular operation and caused my program to crash, hope someone can give me some advice

closed 2024-11-30 by HTRamsey