mavlink/qgroundcontrol #10802: Build problems for android 64 bit(ABIs: arm64-v8a)

RONDODND, 2023-09-07:
When choosing Arm64-v8a, there are build errors that I have given below. When building a 32-bit application, these errors do not occur and everything starts normally.
windows 10, qt 5.15.2, qgc version v4.2.6

linking libQGroundControl_arm64-v8a.so 
ld: error: relocation R_AARCH64_ADR_PREL_PG_HI21 cannot be used against symbol ff_cos_32; recompile with -fPIC
>>> defined in C:/Users/Nikola-PC/mqg_std/qgroundcontrol/src/VideoReceiver/../../gstreamer-1.0-android-universal-1.18.5/arm64/lib\libavcodec.a(fft_float.o)
>>> referenced by asm.S:491 (./libavutil/aarch64\asm.S:491)
>>>               fft_neon.o:(fft32_neon) in archive C:/Users/Nikola-PC/mqg_std/qgroundcontrol/src/VideoReceiver/../../gstreamer-1.0-android-universal-1.18.5/arm64/lib\libavcodec.a

ld: error: can't create dynamic relocation R_AARCH64_ADD_ABS_LO12_NC against symbol: ff_cos_32 in readonly segment; recompile object files with -fPIC or pass '-Wl,-z,notext' to allow text relocations in the output
>>> defined in C:/Users/Nikola-PC/mqg_std/qgroundcontrol/src/VideoReceiver/../../gstreamer-1.0-android-universal-1.18.5/arm64/lib\libavcodec.a(fft_float.o)
>>> referenced by asm.S:491 (./libavutil/aarch64\asm.S:491)
>>>               fft_neon.o:(fft32_neon) in archive C:/Users/Nikola-PC/mqg_std/qgroundcontrol/src/VideoReceiver/../../gstreamer-1.0-android-universal-1.18.5/arm64/lib\libavcodec.a

ld: error: relocation R_AARCH64_ADR_PREL_PG_HI21 cannot be used against symbol ff_cos_64; recompile with -fPIC
>>> defined in C:/Users/Nikola-PC/mqg_std/qgroundcontrol/src/VideoReceiver/../../gstreamer-1.0-android-universal-1.18.5/arm64/lib\libavcodec.a(fft_float.o)
>>> referenced by asm.S:492 (./libavutil/aarch64\asm.S:492)
>>>               fft_neon.o:(fft64_neon) in archive C:/Users/Nikola-PC/mqg_std/qgroundcontrol/src/VideoReceiver/../../gstreamer-1.0-android-universal-1.18.5/arm64/lib\libavcodec.a

ld: error: can't create dynamic relocation R_AARCH64_ADD_ABS_LO12_NC against symbol: ff_cos_64 in readonly segment; recompile object files with -fPIC or pass '-Wl,-z,notext' to allow text relocations in the output
>>> defined in C:/Users/Nikola-PC/mqg_std/qgroundcontrol/src/VideoReceiver/../../gstreamer-1.0-android-universal-1.18.5/arm64/lib\libavcodec.a(fft_float.o)
>>> referenced by asm.S:492 (./libavutil/aarch64\asm.S:492)
>>>               fft_neon.o:(fft64_neon) in archive C:/Users/Nikola-PC/mqg_std/qgroundcontrol/src/VideoReceiver/../../gstreamer-1.0-android-universal-1.18.5/arm64/lib\libavcodec.a

ld: error: relocation R_AARCH64_ADR_PREL_PG_HI21 cannot be used against symbol ff_cos_128; recompile with -fPIC
>>> defined in C:/Users/Nikola-PC/mqg_std/qgroundcontrol/src/VideoReceiver/../../gstreamer-1.0-android-universal-1.18.5/arm64/lib\libavcodec.a(fft_float.o)
>>> referenced by asm.S:493 (./libavutil/aarch64\asm.S:493)
>>>               fft_neon.o:(fft128_neon) in archive C:/Users/Nikola-PC/mqg_std/qgroundcontrol/src/VideoReceiver/../../gstreamer-1.0-android-universal-1.18.5/arm64/lib\libavcodec.a

ld: error: can't create dynamic relocation R_AARCH64_ADD_ABS_LO12_NC against symbol: ff_cos_128 in readonly segment; recompile object files with -fPIC or pass '-Wl,-z,notext' to allow text relocations in the output
>>> defined in C:/Users/Nikola-PC/mqg_std/qgroundcontrol/src/VideoReceiver/../../gstreamer-1.0-android-universal-1.18.5/arm64/lib\libavcodec.a(fft_float.o)
>>> referenced by asm.S:493 (./libavutil/aarch64\asm.S:493)
>>>               fft_neon.o:(fft128_neon) in archive C:/Users/Nikola-PC/mqg_std/qgroundcontrol/src/VideoReceiver/../../gstreamer-1.0-android-universal-1.18.5/arm64/lib\libavcodec.a

ld: error: relocation R_AARCH64_ADR_PREL_PG_HI21 cannot be used against symbol ff_cos_256; recompile with -fPIC
>>> defined in C:/Users/Nikola-PC/mqg_std/qgroundcontrol/src/VideoReceiver/../../gstreamer-1.0-android-universal-1.18.5/arm64/lib\libavcodec.a(fft_float.o)
>>> referenced by asm.S:494 (./libavutil/aarch64\asm.S:494)
>>>               fft_neon.o:(fft256_neon) in archive C:/Users/Nikola-PC/mqg_std/qgroundcontrol/src/VideoReceiver/../../gstreamer-1.0-android-universal-1.18.5/arm64/lib\libavcodec.a

ld: error: can't create dynamic relocation R_AARCH64_ADD_ABS_LO12_NC against symbol: ff_cos_256 in readonly segment; recompile object files with -fPIC or pass '-Wl,-z,notext' to allow text relocations in the output
>>> defined in C:/Users/Nikola-PC/mqg_std/qgroundcontrol/src/VideoReceiver/../../gstreamer-1.0-android-universal-1.18.5/arm64/lib\libavcodec.a(fft_float.o)
>>> referenced by asm.S:494 (./libavutil/aarch64\asm.S:494)
>>>               fft_neon.o:(fft256_neon) in archive C:/Users/Nikola-PC/mqg_std/qgroundcontrol/src/VideoReceiver/../../gstreamer-1.0-android-universal-1.18.5/arm64/lib\libavcodec.a

ld: error: relocation R_AARCH64_ADR_PREL_PG_HI21 cannot be used against symbol ff_cos_512; recompile with -fPIC
>>> defined in C:/Users/Nikola-PC/mqg_std/qgroundcontrol/src/VideoReceiver/../../gstreamer-1.0-android-universal-1.18.5/arm64/lib\libavcodec.a(fft_float.o)
>>> referenced by asm.S:495 (./libavutil/aarch64\asm.S:495)
>>>               fft_neon.o:(fft512_neon) in archive C:/Users/Nikola-PC/mqg_std/qgroundcontrol/src/VideoReceiver/../../gstreamer-1.0-android-universal-1.18.5/arm64/lib\libavcodec.a

ld: error: can't create dynamic relocation R_AARCH64_ADD_ABS_LO12_NC against symbol: ff_cos_512 in readonly segment; recompile object files with -fPIC or pass '-Wl,-z,notext' to allow text relocations in the output
>>> defined in C:/Users/Nikola-PC/mqg_std/qgroundcontrol/src/VideoReceiver/../../gstreamer-1.0-android-universal-1.18.5/arm64/lib\libavcodec.a(fft_float.o)
>>> referenced by asm.S:495 (./libavutil/aarch64\asm.S:495)
>>>               fft_neon.o:(fft512_neon) in archive C:/Users/Nikola-PC/mqg_std/qgroundcontrol/src/VideoReceiver/../../gstreamer-1.0-android-universal-1.18.5/arm64/lib\libavcodec.a

ld: error: relocation R_AARCH64_ADR_PREL_PG_HI21 cannot be used against symbol ff_cos_1024; recompile with -fPIC
>>> defined in C:/Users/Nikola-PC/mqg_std/qgroundcontrol/src/VideoReceiver/../../gstreamer-1.0-android-universal-1.18.5/arm64/lib\libavcodec.a(fft_float.o)
>>> referenced by asm.S:496 (./libavutil/aarch64\asm.S:496)
>>>               fft_neon.o:(fft1024_neon) in archive C:/Users/Nikola-PC/mqg_std/qgroundcontrol/src/VideoReceiver/../../gstreamer-1.0-android-universal-1.18.5/arm64/lib\libavcodec.a

ld: error: can't create dynamic relocation R_AARCH64_ADD_ABS_LO12_NC against symbol: ff_cos_1024 in readonly segment; recompile object files with -fPIC or pass '-Wl,-z,notext' to allow text relocations in the output
>>> defined in C:/Users/Nikola-PC/mqg_std/qgroundcontrol/src/VideoReceiver/../../gstreamer-1.0-android-universal-1.18.5/arm64/lib\libavcodec.a(fft_float.o)
>>> referenced by asm.S:496 (./libavutil/aarch64\asm.S:496)
>>>               fft_neon.o:(fft1024_neon) in archive C:/Users/Nikola-PC/mqg_std/qgroundcontrol/src/VideoReceiver/../../gstreamer-1.0-android-universal-1.18.5/arm64/lib\libavcodec.a

ld: error: relocation R_AARCH64_ADR_PREL_PG_HI21 cannot be used against symbol ff_cos_2048; recompile with -fPIC
>>> defined in C:/Users/Nikola-PC/mqg_std/qgroundcontrol/src/VideoReceiver/../../gstreamer-1.0-android-universal-1.18.5/arm64/lib\libavcodec.a(fft_float.o)
>>> referenced by asm.S:497 (./libavutil/aarch64\asm.S:497)
>>>               fft_neon.o:(fft2048_neon) in archive C:/Users/Nikola-PC/mqg_std/qgroundcontrol/src/VideoReceiver/../../gstreamer-1.0-android-universal-1.18.5/arm64/lib\libavcodec.a

ld: error: can't create dynamic relocation R_AARCH64_ADD_ABS_LO12_NC against symbol: ff_cos_2048 in readonly segment; recompile object files with -fPIC or pass '-Wl,-z,notext' to allow text relocations in the output
[... 12 lines cut ...]
>>>               fft_neon.o:(fft4096_neon) in archive C:/Users/Nikola-PC/mqg_std/qgroundcontrol/src/VideoReceiver/../../gstreamer-1.0-android-universal-1.18.5/arm64/lib\libavcodec.a

ld: error: relocation R_AARCH64_ADR_PREL_PG_HI21 cannot be used against symbol ff_cos_8192; recompile with -fPIC
>>> defined in C:/Users/Nikola-PC/mqg_std/qgroundcontrol/src/VideoReceiver/../../gstreamer-1.0-android-universal-1.18.5/arm64/lib\libavcodec.a(fft_float.o)
>>> referenced by asm.S:499 (./libavutil/aarch64\asm.S:499)
>>>               fft_neon.o:(fft8192_neon) in archive C:/Users/Nikola-PC/mqg_std/qgroundcontrol/src/VideoReceiver/../../gstreamer-1.0-android-universal-1.18.5/arm64/lib\libavcodec.a

[... cut ...]
>>> defined in C:/Users/Nikola-PC/mqg_std/qgroundcontrol/src/VideoReceiver/../../gstreamer-1.0-android-universal-1.18.5/arm64/lib\libavcodec.a(fft_float.o)
>>> referenced by asm.S:500 (./libavutil/aarch64\asm.S:500)
>>>               fft_neon.o:(fft16384_neon) in archive C:/Users/Nikola-PC/mqg_std/qgroundcontrol/src/VideoReceiver/../../gstreamer-1.0-android-universal-1.18.5/arm64/lib\libavcodec.a

ld: error: too many errors emitted, stopping now (use -error-limit=0 to see all errors)
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [Makefile:1962: libQGroundControl_arm64-v8a.so] Error 1

julianoes, 2024-01-09:
I have the same problem in #10949.

DonLakeFlyer, 2024-01-09:
> I have the same problem in #10949.

Huh? #10949 built fine in CI.

julianoes, 2024-01-09:
Not for Android @DonLakeFlyer. It works on my machine for some reason though.

julianoes, 2024-01-09:
https://github.com/mavlink/qgroundcontrol/actions/runs/7467142441/job/20320015104?pr=10949#step:11:1224

DonLakeFlyer, 2024-01-09:
Have you looked at what this talks about: https://stackoverflow.com/questions/13812185/how-to-recompile-with-fpic?

julianoes, 2024-01-09:
Thanks, that looks highly relevant! The thing is that we are not building the gstreamer libs but are using the downloads.

julianoes, 2024-01-09:
Ok, I managed to reproduce the problem locally using NDK 25c instead of NDK21e, using clang-14.

julianoes, 2024-01-10:
@DonLakeFlyer would it be ok to go back to NDK21e? Or should we start building the gstreamer libs ourselves instead?

DonLakeFlyer, 2024-01-10:
Qt6 says: "The required version of the Android NDK for 6.6 is 25.1.8937393."

julianoes, 2024-01-10:
I see. I'll look into building the gstreamer libs then.

julianoes, 2024-01-10:
Looks like it's an ffmpeg (libavutil) issue:
https://issuetracker.google.com/issues/187786235?pli=1

Chromium patched it for them:
https://chromium.googlesource.com/chromium/third_party/ffmpeg/+/67e9e0337185f05a0d8d362473e63d8a9dfdebed

DonLakeFlyer, 2024-01-11:
So what are our options?

julianoes, 2024-01-11:
This fixed it: https://github.com/mavlink/qgroundcontrol/pull/10949/commits/fd003602b9cce6afbc05ca8fa2977444c66bd570

closed 2024-05-31 by HTRamsey