mavlink/qgroundcontrol #10806: Cannot write file /project/build/qmake_qmake_qm_files.qrc: Permission denied

mw66, 2023-09-12:
## Expected Behavior

I'm trying to build using Containers on Linux
```
docker run --rm -v ${PWD}:/project/source -v ${PWD}/build:/project/build qgc-linux-docker
```
run successfully.

## Current Behavior
```
/my/local/home/dir/qgroundcontrol$ docker run --rm -v ${PWD}:/project/source -v ${PWD}/build:/project/build qgc-linux-docker
Info: creating stash file /project/build/.qmake.stash
/opt/Qt/5.15.2/gcc_64/mkspecs/features/toolchain.prf:73: Cannot write cache file /project/build/.qmake.stash: Permission denied
Info: creating stash file /project/build/.qmake.stash
/opt/Qt/5.15.2/gcc_64/mkspecs/features/toolchain.prf:73: Cannot write cache file /project/build/.qmake.stash: Permission denied
Info: creating stash file /project/build/.qmake.stash
/opt/Qt/5.15.2/gcc_64/mkspecs/features/toolchain.prf:73: Cannot write cache file /project/build/.qmake.stash: Permission denied
Info: creating stash file /project/build/.qmake.stash
/opt/Qt/5.15.2/gcc_64/mkspecs/features/toolchain.prf:73: Cannot write cache file /project/build/.qmake.stash: Permission denied
Info: creating stash file /project/build/.qmake.stash
/opt/Qt/5.15.2/gcc_64/mkspecs/features/toolchain.prf:76: Cannot write cache file /project/build/.qmake.stash: Permission denied
Info: creating stash file /project/build/.qmake.stash
/opt/Qt/5.15.2/gcc_64/mkspecs/features/toolchain.prf:409: Cannot write cache file /project/build/.qmake.stash: Permission denied
Info: creating stash file /project/build/.qmake.stash
/opt/Qt/5.15.2/gcc_64/mkspecs/features/toolchain.prf:411: Cannot write cache file /project/build/.qmake.stash: Permission denied
Project MESSAGE: Qt version 5.15.2
Project MESSAGE: Linux build
/usr/bin/ccache
Project MESSAGE: Found ccache, enabling
Project MESSAGE: Daily Build
Project MESSAGE: GIT_DESCRIBE v4.2.0-471-gffda931df
Project MESSAGE: QGroundControl APP_VERSION_STR VERSION v4.2.0-471-gffda931df 4.2.0.471
Project MESSAGE: Release flavor
Project MESSAGE: Skipping support for Pairing
Project MESSAGE: Using MAVLink dialect 'all'.
Project MESSAGE: Skipping support for Zeroconf (unsupported platform)
Project MESSAGE: Taisync disabled
Project MESSAGE: Microhard disabled
Project MESSAGE: Skipping support for video streaming (manual override from command line)
collect2 version 9.4.0
/usr/bin/ld.gold -plugin /usr/lib/gcc/x86_64-linux-gnu/9/liblto_plugin.so -plugin-opt=/usr/lib/gcc/x86_64-linux-gnu/9/lto-wrapper -plugin-opt=-fresolution=/tmp/ccxFnIm6.res -plugin-opt=-pass-through=-lgcc_s -plugin-opt=-pass-through=-lgcc -plugin-opt=-pass-through=-lc -plugin-opt=-pass-through=-lgcc_s -plugin-opt=-pass-through=-lgcc --build-id --eh-frame-hdr -m elf_x86_64 --hash-style=gnu --as-needed -dynamic-linker /lib64/ld-linux-x86-64.so.2 -pie -z now -z relro /usr/lib/gcc/x86_64-linux-gnu/9/../../../x86_64-linux-gnu/Scrt1.o /usr/lib/gcc/x86_64-linux-gnu/9/../../../x86_64-linux-gnu/crti.o /usr/lib/gcc/x86_64-linux-gnu/9/crtbeginS.o -L/usr/lib/gcc/x86_64-linux-gnu/9 -L/usr/lib/gcc/x86_64-linux-gnu/9/../../../x86_64-linux-gnu -L/usr/lib/gcc/x86_64-linux-gnu/9/../../../../lib -L/lib/x86_64-linux-gnu -L/lib/../lib -L/usr/lib/x86_64-linux-gnu -L/usr/lib/../lib -L/usr/lib/gcc/x86_64-linux-gnu/9/../../.. --version -lstdc++ -lm -lgcc_s -lgcc -lc -lgcc_s -lgcc /usr/lib/gcc/x86_64-linux-gnu/9/crtendS.o /usr/lib/gcc/x86_64-linux-gnu/9/../../../x86_64-linux-gnu/crtn.o
GNU gold (GNU Binutils for Ubuntu 2.34) 1.16
Copyright (C) 2020 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or (at your option) a later version.
This program has absolutely no warranty.
/opt/Qt/5.15.2/gcc_64/mkspecs/features/resources_functions.prf:80: Cannot write file /project/build/qmake_qmake_qm_files.qrc: Permission denied
```

## Steps to Reproduce:
I followed exact step on:

https://dev.qgroundcontrol.com/master/en/getting_started/container.html

## System Information
When posting bug reports, include the following information
- Operating System: Ubuntu 22.04 LTS
- QGC Version: current git clone

mw66, 2023-09-12:
I tried:

$ docker run -it  qgc-linux-docker sh

then:
```
$ ls -l /project
total 4
drwxr-xr-x 2 root root 4096 Sep 12 00:49 build

$ whoami
user
```

so `user` does not have write permission to `/project/build` ?

mw66, 2023-09-12:
I build as root in that image, add `-u 0`

```
$ docker run -u 0 --rm -v ${PWD}:/project/source -v ${PWD}/build:/project/build qgc-linux-docker
```

seems working so far.

mw66, 2023-09-12:
I think the real fix should be `chown user /project/build`

yi452873250, 2024-01-29:
I meet this problem：
run code:
sudo docker build --file ./deploy/docker/Dockerfile-build-linux -t qgc-linux-docker .
show success

[image]
but run:
 sudo docker run --rm -v ${PWD}:/project/source -v ${PWD}/build:/project/build qgc-linux-docker
[image]
please,how can I do？

evigdorchik, 2024-02-21:
@yi452873250 I experienced the same thing as you. My solution was to checkout the 4.3.0 tag and the run `mkdir build
docker run --rm -v ${PWD}:/project/source -v ${PWD}/build:/project/build qgc-linux-docker` again.
Hope that helps?