mavlink/qgroundcontrol #10807: undefined symbol: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEED1Ev, version Qt_5

mw66, 2023-09-12:
## Expected Behavior

I'm trying to build using Containers on Linux, after the build,

it should run successfully.

## Current Behavior
```
$ cd build/
$ ./staging/QGroundControl 
./staging/QGroundControl: symbol lookup error: ./staging/QGroundControl: undefined symbol: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEED1Ev, version Qt_5
```

## Steps to Reproduce:
I followed exact step on:

https://dev.qgroundcontrol.com/master/en/getting_started/container.html

## System Information
When posting bug reports, include the following information
- Operating System: Ubuntu 22.04 LTS
- QGC Version: current git clone

mw66, 2023-09-12:
related issue: https://github.com/mavlink/qgroundcontrol/issues/10578

RoundPegInASquareHole, 2024-05-05:
for me, executing the file build/staging/QGroundControl with double click worked haha (i am using the version 4.2.9)

Alex-Beh, 2024-05-07:
@RoundPegInASquareHole same for me also.

closed 2024-11-30 by HTRamsey