mavlink/qgroundcontrol #10809: What is the key / the process to choose commits for stable version?

zdanek, 2023-09-13:
Hi.
This seems a question mostly for @patrickelectric 

I try to merge my custom changes with latest stable. I took it from Stable_V4.2
I tried to cherry-pick some necessary commits from master. But I found some problems with missing other code. This is not surprising. What surprised me was:
* commits on stable branch are in "random" order in regard of the date 
* the policy of choosing commits to be taken into stable is unknown for me

Regarding order of commits: on master, I see, they are also switched. Does it mean, git takes original commit date, not merge / rebase date?

Regarding "policy": how do you choose what to take? Is it like take everything from master up to particular day? Or picked by some key?

Thanks :)

closed 2024-04-05 by HTRamsey

patrickelectric, 2024-04-06:
Hi @HTRamsey! Sorry for not replying it early, I usually merge what people ask to get merge vs what is feasible and safe to gets merged. Also PRs against the stable also help. It's hard to keep track of everything. Currently I'm a bit off the project. But I hope that it answer your question.

HTRamsey, 2024-04-06:
Oh you're fine, I figured if @zdanek still had the question 6 months later he would just ask on Discord or something so I went ahead and closed it. I'm trying to help clean up the issues a little bit but it's much much worse than I thought haha