mavlink/qgroundcontrol #10811: Richtext message display broken

bkueng, 2023-09-13:
It looks like https://github.com/mavlink/qgroundcontrol/pull/10603 broke rich text rendering:

[image]

[image]

@DonLakeFlyer could you have a look?

DonLakeFlyer, 2023-09-20:
Broken by this as well: https://github.com/mavlink/qgroundcontrol/pull/10533

DonLakeFlyer, 2023-09-20:
To me using formatting in status text seems like a not so great idea. Not sure how you keep both text like this "foo < bar" which could be in any status text message and this "<b>foo</b>" working correctly.

bkueng, 2023-09-21:
The above comes through the events interface: https://github.com/mavlink/qgroundcontrol/blob/master/src/Vehicle/Vehicle.cc#L1623.
And the escaping is missing there as well. Initially I just did it that way to fit into the existing UI elements but ideally we would do a bit more.

DonLakeFlyer, 2023-09-22:
OK, so can you fix this then? The vehicle message stuff was never meant to understand how to do rich text from the vehicle itself. It only used rich text for internal QGC formatting usage.

You'll need to escape the regular text coming through the event stuff. And then I would hack UASMessageHandler for now to look for a "<br/>" in the message and skip the escaping there if so. And then add rich text format back to the critical message label in MainRootWindow. I started to hack it myself, but I'm not quite so sure about that event code to get it right. I also couldn't get the failsafe to generate an event to test it with.

closed 2023-09-27 by DonLakeFlyer