mavlink/qgroundcontrol #10812: Android Release APK unsafe installation

fraviofii, 2023-09-18:
I have a QGC project fork and I intend to create an APK file to be installed on my tablet.

I had the problem with `ANDROID_KEYSTORE_PASSWORD`, which I was able to solve using the instructions provide on this link: https://github.com/mavlink/qgroundcontrol/pull/10398.

I can have a built APK. But, when I copy the file to my tablet (Samsung Galaxy A8), I have the message saying that it is an *unsafe* installation.

But, when I download the Release version of QCG from Github and use the same installation approach, I do not have such behavior.

What should I to to have a *not unsafe* QGC APK from my fork?

danieffe, 2023-11-15:
Excuse me, how did you build the apk? Did you use Android Studio? What file in the main repository is responsable of the building process for Android? (I'm using the comunity container Docker on Ubuntu 22.04). I'm still a newbie, but I created my custom version of QGC and I wanted to see all the changes that I made in the Android app FlyView. Thanks

HTRamsey, 2024-05-31:
It must be a signed release build

closed 2024-05-31 by HTRamsey