mavlink/qgroundcontrol #10814: RC Failsafe warning not showing up

ryanjAA, 2023-09-21:
On px4 1.14 and QGC daily. 

Make RC failsafe happen and the vehicle does what it is supposed to but the text on the screen doesn’t show up. Grabbed an old 4.1.6 QGC version (just because it was already downloaded) and tried on that and it worked fine with the RC loss message showing up. 

I have seen work on versions circa 4.2.4 so not sure exactly where it died but not working on daily or release.

DonLakeFlyer, 2023-09-22:
Did it stop working because of new QGC or new firmware? Wondering if firmware is marking severity wrong. Does the message show up in the messages dropdown? Is it red colored in there?

ryanjAA, 2023-09-23:
I think QGC because I can keep the firmware the same, go back to something old like QGC 4.1.6 and it shows up. 

Doesn’t show up in the drop down (so not red or anything) but firmware is putting it in the correct mode (aka if return is set, it’s going to return). Also doing this with datalink failsafe.

DonLakeFlyer, 2023-09-23:
@ryanjAA Can you save telelemetry log and attach so I can play this back to see what is happening.

DonLakeFlyer, 2023-09-28:
Can you try this again with lastest daily. Beat Keung cleaned up status text stuff that may have been broken. Merged in a couple days back

ryanjAA, 2023-09-28:
I totally forgot to send you the tlogs. Testing now. Apologies.

closed 2026-10-04 by github-actions[bot]