mavlink/qgroundcontrol #10825: Can't build Stable_V4.2 QGroundControl, nlohman_json not found

iura1, 2023-10-03:
## Expected Behavior
Build and run Stable_V4.2 QGroundControl version.

## Current Behavior
After I pull the code and checkout the Stable_V4.2 branch, I run again: git submodule update --recursive, but I get this error:
fatal: remote error: upload-pack: not our ref 391786c6c3abdd3eeb993a3154f1f2a4cfe137a0
fatal: Fetched in submodule path 'libs/libevents/libevents/libs/cpp/parse/nlohmann_json', but it did not contain 391786c6c3abdd3eeb993a3154f1f2a4cfe137a0. Direct fetching of that commit failed.
fatal: Failed to recurse into submodule path 'libs/libevents/libevents'

Also if I build the code from QtCreator It is failing with nlohman_json not found error.
It is strange because several weeks ago I pulled the code from the same branch Stable_V4.2 and it worked.

## Steps to Reproduce:
1. git clone --recursive -j8 https://github.com/mavlink/qgroundcontrol.git
2. git submodule update --recursive
3. git checkout -f Stable_V4.2
4. git submodule update --recursive
5. open qgroundcontrol.pro in QT Creator 
6. wait for indexing
7. build

## System Information
- Operating System: Windows 10
- QGC Version: V4.2
- QT Creator 7.0.2
- Desktop Qt 5.15.2 MSVC2019 64bit Debug

## Detailed Description
[image]

codymarquardt, 2023-10-14:
Hey, I just had this problem. I am currently struggling with the exact same setup and the error that the "command line is too long", but I figured out your issue for myself. 

If you go to /libs/libevents/libevents/ and run git submodule init && git submodule update, that will pull the latest libevents that has the file you are missing

AndKe, 2023-10-19:
@DonLakeFlyer   same issue on Linux, it's been like that for a while now.
deinit/reinit submodules fail like this:
```
$ git submodule update --recursive
fatal: remote error: upload-pack: not our ref bc889afb4c5bf1c0d8ee29ef35eaaf4c8bef8a5d
fatal: Fetched in submodule path 'libs/libevents/libevents/libs/cpp/parse/nlohmann_json', but it did not contain bc889afb4c5bf1c0d8ee29ef35eaaf4c8bef8a5d. Direct fetching of that commit failed.
fatal: Failed to recurse into submodule path 'libs/libevents/libevents'

```

and compilation fails like this:
```
$ make -j20
compiling /home/andre/prog/qgroundcontrol/src/QtLocationPlugin/QGCMapEngine.cpp
compiling /home/andre/prog/qgroundcontrol/src/QtLocationPlugin/QGCMapUrlEngine.cpp
compiling /home/andre/prog/qgroundcontrol/src/QtLocationPlugin/MapProvider.cpp
compiling /home/andre/prog/qgroundcontrol/src/QtLocationPlugin/ElevationMapProvider.cpp
compiling /home/andre/prog/qgroundcontrol/src/QtLocationPlugin/GoogleMapProvider.cpp
compiling /home/andre/prog/qgroundcontrol/src/QtLocationPlugin/BingMapProvider.cpp
compiling /home/andre/prog/qgroundcontrol/src/QtLocationPlugin/GenericMapProvider.cpp
compiling /home/andre/prog/qgroundcontrol/src/QtLocationPlugin/EsriMapProvider.cpp
compiling /home/andre/prog/qgroundcontrol/src/QtLocationPlugin/MapboxMapProvider.cpp
compiling /home/andre/prog/qgroundcontrol/libs/libevents/libevents/libs/cpp/protocol/receive.cpp
compiling /home/andre/prog/qgroundcontrol/libs/libevents/libevents/libs/cpp/parse/health_and_arming_checks.cpp
compiling /home/andre/prog/qgroundcontrol/libs/libevents/libevents/libs/cpp/parse/parser.cpp
In file included from /home/andre/prog/qgroundcontrol/libs/libevents/libevents/libs/cpp/parse/health_and_arming_checks.h:8,
                 from /home/andre/prog/qgroundcontrol/libs/libevents/libevents/libs/cpp/parse/health_and_arming_checks.cpp:1:
/home/andre/prog/qgroundcontrol/libs/libevents/libevents/libs/cpp/parse/parser.h:13:10: fatal error: nlohmann_json/single_include/nlohmann/json_fwd.hpp: No such file or directory
   13 | #include "nlohmann_json/single_include/nlohmann/json_fwd.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.

```

peterbarker, 2023-10-24:
@bkueng just a FYI - libevents is referencing a commit which doesn't exist in the repository referenced in the remote for `nlohmann_json` and this is killing qgroundcontrol builds.  I'm guessing many devs have that as an extra remote.

For others who need a workaround 'til things get fixed, try:
```
cd libs/libevents/libevents/libs/cpp/parse/nlohmann_json
 git remote add nlohmann https://github.com/nlohmann/json
git fetch nlohmann
cd -
git submodule update --init --recursive
```

That should indicate the ref is now present in the subrepository.

Retrying your build should work out better.

AndKe, 2023-10-24:
What fixed the issue for me, was to enter each of the directories for the failing nubmodule, and doing ```git reset && git checkout . ```

DonLakeFlyer, 2023-10-30:
So what's the story here? I can't quite figure out what the current state is? Is this only a Stable problem or is it a master problem? Or is it only a problem if you switch between stable and master? Does someone already have a change that needs to get into the repo to fix it?

AndKe, 2023-10-30:
@DonLakeFlyer I think you can close the issue, this is somehow something that happens now-and-then (maybe yearly for me) that submodules somehow go bad. 
I do not know how/why .  Ardupilot project has also a lot of submodules, but they never go bad.
FWIW: this is not a issue in the master.

DonLakeFlyer, 2023-10-30:
Thanks @andke. That's kind of what I thought since I've had similar things happen to me. For me it tends to be triggered by going far back into older commits and then trying to get back to head. Git just seems to get confused.

DonLakeFlyer, 2023-11-13:
I ran into this like three times in the last week. Had to use @peterbarker fix to get going again. So depending on what you are doing it can pop up a lot. @bkueng Do you know how to fix this?

bkueng, 2023-11-14:
Yes git isn't handling the change very well. This should work:
```
git submodule sync --recursive
git submodule update --recursive 
```

DonLakeFlyer, 2023-11-14:
Just to make sure: So this is just git insanity. The repo is set up correctly. I'll close this again.