mavlink/qgroundcontrol #10827: Tutorial for adding a new module to the overall project

ascoril, 2023-10-11:
Hi,

I could not find in the developer guide steps for a beginner to add a new module which can be referenced from the PlanView.qml file. Could anyone help with the steps or add that to the developer guide documentation?

Best

# This system is for posting bugs or feature requests ONLY. Support questions will be automatically closed with no response.
# For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources

----
# To post a bug report
When posting bug reports, include the following informaiton:
- Ground station operating system.
- QGroundControl version and build (daily, stable, self-built from source, etc.)
- **If you are using a Stable build which is not the latest, first step is to install latest Stable build and try again prior to entering a bug report.**
- Autopilot board: Pixhawk I, Pixhawk Mini, Pixhawk 2, etc.
- Autopilot firmware (e.g. PX4, ArduPilot, custom) and version.
- Exact steps to reproduce the problem. Starting from booting QGroundControl to when the problem occurs.

**All this template text should be deleted before creating your issue.**

JMare, 2023-10-17:
Hello,

Assuming you want to add your own C++ class which can be accessed from the QML, I would suggest the following:

Setup a custom build with your own CorePlugin

Define your class XYZ in its own cc/h files, make sure it is a Q Object and use properties to expose anything you want to.

Make the object of class XYZ a member of the core plugin. Provide access to this type via a QProperty, eg
- Q_PROPERTY(XYZ*   xyz  MEMBER xyz CONSTANT);
- XYZ* _xyz = nullptr;
- Create function XYZ* xyz(void) which the property will call.

In the core plugin constructor, register XYZ as a qml type, and create the actual object
- qmlRegisterUncreatableType<XYZ>("QGroundControl.XYZ", 1, 0, "XYZ", "Reference only");
- _xyz = new XYZ(this);

You should find you are now able to import QGroundcontrol.corePlugin.xyz and then access properties in qml
In the core plugin settoolbox function, you can link signals to slots in xyz. eg:
-     connect(toolbox->mavlinkProtocol(), &MAVLinkProtocol::messageReceived, _xyz, &XYZ::handleMavlinkMessage);

This is just a rough list of what I did in my own custom build.
I found that making the custom class a member of the core plugin, I was able to do everything I needed without changing too much of the upstream code.
The alternative is to make your custom class a standalone tool in the qgc toolbox, but that requires touching more upstream files.
For examples of how this works you could look at the toolbox code and see how things like the microhard modules have been added.

Hope this is helpful.

jorgitinlugo, 2024-12-20:
Is there any example i can see to know how to do it?

closed 2026-10-04 by github-actions[bot]