mavlink/qgroundcontrol #10830: Cannot compile QGroundControl on Windows

codymarquardt, 2023-10-14:
## Expected Behavior
I am trying to build QgroundControl from source on windows. When I build the code, I expect it to work successfully

## Current Behavior
I receive an error that "the command line is too long"

## Steps to Reproduce:
1. pull Stablev4.2 of qgroundcontrol
2. install qt using online installer
3. open project in qtcreator using the .pro file
4. press the hammer/build button in qtcreator

## System Information
- Operating System: Windows 11
- QGC Version: Stable V4.2
- QGC build: Stable V4.2
- Flight Controller: Pixhawk I
- Autopilot (with version): ArduSub 4.0

## Log Files and Screenshots

[image]
[image]

dogmaphobic, 2023-10-31:
I'm not an expert on Windows dev. I wonder if there is some setting to extend the command line length limit you see above. Try moving the QGC source to a shorter path (instead of `C:\Users\codym\OneDrive - Texas A&M University\BlueROVCodeDevelopment\qgroundcontrol`.)

codymarquardt, 2023-11-02:
I moved to a path that was shorter and without spaces. I am unsure which was the fix, but it resolved the issue. Thanks!

closed 2024-05-31 by HTRamsey