mavlink/qgroundcontrol #10860: QGroundControl no runing in Linux

SebastianViollaz, 2023-11-17:
## Expected Behavior
Make an script in python that open the QGroundControl.AppImage.

[image]
This script should run the app ( stable 4.28 )
## Current Behavior

QSocketNotifier: Can only be used with threads started with QThread
Settings location "/home/seba/.config/QGroundControl.org/QGroundControl.ini" Is writable?: true
Filter rules "*Log.debug=false\nGStreamerAPILog.debug=true\nqt.qml.connections=false"
System reported locale: QLocale(English, Latin, United States) ; Name "en_US" ; Preffered (used in maps):  "en-US"
serialnmea: No known GPS device found. Specify the COM port via QT_NMEA_SERIAL_PORT.
VideoReceiverLog: Stop called on empty URI
VideoReceiverLog: Stop called on empty URI
MAVLinkLogManagerLog: MAVLink logs directory: "/home/seba/Documents/Dron/Software/QGC/Fuentes/PruebasAutoCargado/Logs"
qt.network.ssl: QSslSocket: cannot resolve EVP_PKEY_base_id
qt.network.ssl: QSslSocket: cannot resolve SSL_get_peer_certificate
Map Cache in: "/home/seba/.cache/QGCMapCache300" / "qgcMapCache.db"
qt.qpa.wayland: Failed to load client buffer integration: "wayland-egl"
qt.qpa.wayland: Available client buffer integrations: ()
qml: QGCCorePlugin(0x558585db1910) []
qt.qpa.wayland: No shell integration named "xdg-shell" found
qt.qpa.wayland: No shell integration named "xdg-shell-v6" found
qt.qpa.wayland: No shell integration named "wl-shell" found
qt.qpa.wayland: No shell integration named "ivi-shell" found
qt.qpa.wayland: Loading shell integration failed.
qt.qpa.wayland: Attempted to load the following shells ("xdg-shell", "xdg-shell-v6", "wl-shell", "ivi-shell")
setCurrentPlanViewSeqNum
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_get_peer_certificate
setCurrentPlanViewSeqNum
_recalcFlightPathSegments homePositionValid false
_recalcFlightPathSegments homePositionValid false
qt.positioning.geoclue2: Unable to start the client: "org.freedesktop.DBus.Error.AccessDenied" "Geolocation disabled for UID 1000"
QGCPositionManager error 0
Download QGC stable version failed "Error during download. Error: 6"

I think the app runs but no GUI is displayed.

## System Information

- Operating System: Ubuntu 22.04.2 LTS
- QGC Version: [e.g. 3.4.4]
- QGC build: 4.28 Stable

zdanek, 2023-11-17:
Might be connected to Wayland maybe. It introduces more safety in display system. Some libs / api calls stop to work. 

Did you try to run as root? It's forbidden with Wayland.

Other idea, probably stupid, try to run with env variable
export GDK_BACKEND=x11

I know it might bad path but I was using this to use old x11 display system instead of new, Wayland.

Give us some feedback on this.

zdanek, 2023-11-17:
Alright, comment above was on my phone. Now I got my hands on pc keyboard :D 

Better ideas:
* try to run with `export DISABLE_WAYLAND=1`
* try to install qtwayland plugin `sudo apt install qtwayland5`

My request to is to try it in order above, that is, check `DISABLE_WAYLAND` first and note the result. I'd like to learn something with this case :)

Give us feedback.

SebastianViollaz, 2023-11-17:
@zdanek Thanks i will try to do that thanks, I will let u know. 

I can run with the command .`/QGroundControl.AppImage` in the actual terminal  (or double clicking) and it works fine. 
I also made an script that execute that command directly in the terminal but same result.
I read that could be something related with the environment variables (CHAT GPT).

In windows this works fine, the problem is with Linux.

When i run my script and then try to open manually the "Can't run a second instance of qgc" msg apears so i supose that it is runing but without the GUI.

zdanek, 2023-11-18:
what about the other  solutions I proposed ?

SebastianViollaz, 2023-11-19:
On Tuesday I will take a look to it

SebastianViollaz, 2023-11-21:
@zdanek I tried to do both things, anything change. Before installing qtwayland5 the output was the same but 3 more msg where printed:
Done
double free or corruption (fasttop)
Aborted (core dumped)

SebastianViollaz, 2023-11-21:
when i run the App Image from the terminal i have this output, (The GUI runs normally).
There are some differences between this and what i get in the console while running my script:

Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
Settings location "/home/seba/.config/QGroundControl.org/QGroundControl.ini" Is writable?: true
Filter rules "*Log.debug=false\nGStreamerAPILog.debug=true\nqt.qml.connections=false"
System reported locale: QLocale(English, Latin, United States) ; Name "en_US" ; Preffered (used in maps):  "en-US"
serialnmea: No known GPS device found. Specify the COM port via QT_NMEA_SERIAL_PORT.
VideoReceiverLog: Stop called on empty URI
VideoReceiverLog: Stop called on empty URI
MAVLinkLogManagerLog: MAVLink logs directory: "/home/seba/Documents/Dron/Software/QGC/Fuentes/PruebasAutoCargado/Logs"
qt.network.ssl: QSslSocket: cannot resolve EVP_PKEY_base_id
qt.network.ssl: QSslSocket: cannot resolve SSL_get_peer_certificate
Map Cache in: "/home/seba/.cache/QGCMapCache300" / "qgcMapCache.db"
qml: QGCCorePlugin(0x55a58cac4ed0) []
setCurrentPlanViewSeqNum
setCurrentPlanViewSeqNum
_recalcFlightPathSegments homePositionValid false
_recalcFlightPathSegments homePositionValid false
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_get_peer_certificate
qt.positioning.geoclue2: Unable to start the client: "org.freedesktop.DBus.Error.AccessDenied" "Geolocation disabled for UID 1000"
QGCPositionManager error 0
Download QGC stable version failed "Error during download. Error: 6"

SebastianViollaz, 2023-11-21:
@zdanek  I make it work. I read about wayland and X11. In my Ubuntu session i change the session type to use Xorg instead of wayland.  Then i run the script and work  properly.

zdanek, 2023-11-21:
Yeah. This was my shot. Not exactly I knew how to express it or configure but this was my guess. I'm having similar problem with python + gstreamer app.

So, it would be beneficial for me and rest of devs for you to share a solution, maybe with some comment / explanation.
Thanks.

SebastianViollaz, 2023-11-21:
Change the System type to x11 change between Linux Distributions,
Im using **Ubuntu**. To change the configuration just logout and before entering the password i search for the settings' gear. The i chose the type i want to use.
[image]
In my case instead of "Ubuntu" i see an option that says "Ubuntu on Xor" and that the one that use x11.

Anyways this isn't a solution  for my case. I'm developing an app that runs in Wayland so i cant use the app if i change this setting.

zdanek, 2023-11-21:
Did you install qtwayland5?

SebastianViollaz, 2023-11-21:
Yes I have it installed.

zdanek, 2023-11-22:
> Yes I have it installed.

So, to recap, you solved problem by choosing x11 display during login, right?

Did you try to use Wayland, send enable it with qt?

SebastianViollaz, 2023-11-22:
Yes i solved choosing x11 in login page. But with wayland instead of x11 i find no solution.
I tried runing with Wayland enabled and disbled, also instaling the qtwayland5.

christianrauch, 2023-11-26:
I am using the flatpak (https://flathub.org/apps/org.mavlink.qgroundcontrol) on Wayland without issues.

Follow the instructions at https://flathub.org/setup/Ubuntu to install flatpak and set up flathub and then:
```
flatpak install flathub org.mavlink.qgroundcontrol
```

zdanek, 2023-11-26:
@SebastianViollaz what I meant, was to also try to run explicitly on Wayland, something like here
https://discourse.ubuntu.com/t/environment-variables-for-wayland-hackers/12750

and probably explicitly setting `$QT_QPA_PLATFORM` but please dig into it.

chutsu, 2024-01-24:
> I am using the flatpak (https://flathub.org/apps/org.mavlink.qgroundcontrol) on Wayland without issues.
> 
> Follow the instructions at https://flathub.org/setup/Ubuntu to install flatpak and set up flathub and then:
> 
> ```
> flatpak install flathub org.mavlink.qgroundcontrol
> ```

Thanks @christianrauch this workaround worked in Ubuntu 22.04 under wayland.

closed 2024-02-03 by zdanek