mavlink/qgroundcontrol #10868: build error on ubuntu20

rebelboy, 2023-11-23:
01:03:13: Running steps for project qgroundcontrol...
01:03:14: Starting: "/home/qgc/Qt/5.15.2/android/bin/qmake" /home/qgc/Downloads/qgroundcontrol/qgroundcontrol.pro -spec android-clang CONFIG+=qtquickcompiler ANDROID_ABIS=arm64-v8a
Project MESSAGE: Qt version 5.15.2
Project MESSAGE: Android Arm 64 bit build
Project MESSAGE: Daily Build
sh: 1: git: not found
sh: 1: git: not found
sh: 1: git: not found
sh: 1: git: not found
Project MESSAGE: GIT_DESCRIBE
Project MESSAGE: QGroundControl APP_VERSION_STR VERSION vUnknown 0.0.0
Project MESSAGE: VERSION 0.0.0
Project MESSAGE: Android version info: 660000000 bitness:66 major:0 minor:0 patch:00 dev:000
Project MESSAGE: Release flavor
Project MESSAGE: Skipping support for Pairing
Project MESSAGE: Using MAVLink dialect 'all'.
Project MESSAGE: Skipping support for Zeroconf (unsupported platform)
Project MESSAGE: Taisync disabled
Project MESSAGE: Microhard disabled
Project MESSAGE: Skipping support for video streaming (GStreamer libraries not installed)
Project MESSAGE: Installation instructions here: https://github.com/mavlink/qgroundcontrol/blob/master/src/VideoReceiver/README.md
Project MESSAGE: This project is using private headers and will therefore be tied to this specific Qt module build version.
Project MESSAGE: Running this project against other versions of the Qt modules may crash at any arbitrary point.
Project MESSAGE: This is not a bug, but a result of using Qt internals. You have been warned!
Project MESSAGE: Qt version 5.15.2
Project MESSAGE: Android Arm 64 bit build
Project MESSAGE: Daily Build
sh: 1: git: not found
sh: 1: git: not found
sh: 1: git: not found
sh: 1: git: not found
Project MESSAGE: GIT_DESCRIBE
Project MESSAGE: QGroundControl APP_VERSION_STR VERSION vUnknown 0.0.0
Project MESSAGE: VERSION 0.0.0
Project MESSAGE: Android version info: 660000000 bitness:66 major:0 minor:0 patch:00 dev:000
Project MESSAGE: Release flavor
Project MESSAGE: Skipping support for Pairing
Project MESSAGE: Using MAVLink dialect 'all'.
Project MESSAGE: Skipping support for Zeroconf (unsupported platform)
Project MESSAGE: Taisync disabled
Project MESSAGE: Microhard disabled
Project MESSAGE: Skipping support for video streaming (GStreamer libraries not installed)
Project MESSAGE: Installation instructions here: https://github.com/mavlink/qgroundcontrol/blob/master/src/VideoReceiver/README.md
01:03:18: The process "/home/qgc/Qt/5.15.2/android/bin/qmake" exited normally.
01:03:18: Starting: "/home/qgc/Android/Sdk/ndk/21.3.6528147/prebuilt/linux-x86_64/bin/make" -f /home/qgc/Downloads/build-qgroundcontrol-Android_Qt_5_15_2_Clang_Multi_Abi-Release/Makefile qmake_all
make: Nothing to be done for 'qmake_all'.
01:03:18: The process "/home/qgc/Android/Sdk/ndk/21.3.6528147/prebuilt/linux-x86_64/bin/make" exited normally.
01:03:18: Starting: "/home/qgc/Android/Sdk/ndk/21.3.6528147/prebuilt/linux-x86_64/bin/make" -j4
rcc qgroundcontrol_qmlcache.qrc
rcc src_FirmwarePlugin_APM_APMResources_qmlcache.qrc
rcc src_FirmwarePlugin_PX4_PX4Resources_qmlcache.qrc
linking libQGroundControl_arm64-v8a.so
echo Post Link Common
Post Link Common
01:03:30: The process "/home/qgc/Android/Sdk/ndk/21.3.6528147/prebuilt/linux-x86_64/bin/make" exited normally.
01:03:30: Removing directory /home/qgc/Downloads/build-qgroundcontrol-Android_Qt_5_15_2_Clang_Multi_Abi-Release/android-build/assets
01:03:30: Removing directory /home/qgc/Downloads/build-qgroundcontrol-Android_Qt_5_15_2_Clang_Multi_Abi-Release/android-build/libs
01:03:30: Starting: "/home/qgc/Android/Sdk/ndk/21.3.6528147/prebuilt/linux-x86_64/bin/make" INSTALL_ROOT=/home/qgc/Downloads/build-qgroundcontrol-Android_Qt_5_15_2_Clang_Multi_Abi-Release/android-build install && cd /home/qgc/Downloads/build-qgroundcontrol-Android_Qt_5_15_2_Clang_Multi_Abi-Release && /home/qgc/Android/Sdk/ndk/21.3.6528147/prebuilt/linux-x86_64/bin/make INSTALL_ROOT=/home/qgc/Downloads/build-qgroundcontrol-Android_Qt_5_15_2_Clang_Multi_Abi-Release/android-build install
/home/qgc/Qt/5.15.2/android/bin/qmake -install qinstall -exe libQGroundControl_arm64-v8a.so /home/qgc/Downloads/build-qgroundcontrol-Android_Qt_5_15_2_Clang_Multi_Abi-Release/android-build/libs/arm64-v8a/libQGroundControl_arm64-v8a.so
/home/qgc/Qt/5.15.2/android/bin/qmake -install qinstall -exe libQGroundControl_arm64-v8a.so /home/qgc/Downloads/build-qgroundcontrol-Android_Qt_5_15_2_Clang_Multi_Abi-Release/android-build/libs/arm64-v8a/libQGroundControl_arm64-v8a.so
01:03:32: The process "/home/qgc/Android/Sdk/ndk/21.3.6528147/prebuilt/linux-x86_64/bin/make" exited normally.
01:03:32: Starting: "/home/qgc/Qt/5.15.2/android/bin/androiddeployqt" --input /home/qgc/Downloads/build-qgroundcontrol-Android_Qt_5_15_2_Clang_Multi_Abi-Release/android-QGroundControl-deployment-settings.json --output /home/qgc/Downloads/build-qgroundcontrol-Android_Qt_5_15_2_Clang_Multi_Abi-Release/android-build --android-platform android-34 --jdk /usr/lib/jvm/java-17-openjdk-amd64 --gradle --release
Generating Android Package
  Input file: /home/qgc/Downloads/build-qgroundcontrol-Android_Qt_5_15_2_Clang_Multi_Abi-Release/android-QGroundControl-deployment-settings.json
  Output directory: /home/qgc/Downloads/build-qgroundcontrol-Android_Qt_5_15_2_Clang_Multi_Abi-Release/android-build/
  Application binary: QGroundControl
  Android build platform: android-34
  Install to device: No
Warning: QML import could not be resolved in any of the import paths: org.freedesktop.gstreamer.GLVideoItem
Warning: QML import could not be resolved in any of the import paths: QGroundControl.Palette
Warning: QML import could not be resolved in any of the import paths: QGroundControl.MultiVehicleManager
Warning: QML import could not be resolved in any of the import paths: QGroundControl.QGCPositionManager
Warning: QML import could not be resolved in any of the import paths: QGroundControl.SettingsManager
Warning: QML import could not be resolved in any of the import paths: QGroundControl.Controllers
Warning: QML import could not be resolved in any of the import paths: QGroundControl.Templates
Warning: QML import could not be resolved in any of the import paths: QGroundControl.ShapeFileHelper
Warning: QML import could not be resolved in any of the import paths: QGroundControl.ScreenToolsController
Warning: QML import could not be resolved in any of the import paths: QGroundControl.QGCMapEngineManager
Warning: QML import could not be resolved in any of the import paths: QGroundControl.ArduPilot
Warning: QML import could not be resolved in any of the import paths: QGroundControl.AutoPilotPlugin
Warning: QML import could not be resolved in any of the import paths: MAVLink
Warning: QML import could not be resolved in any of the import paths: Custom.Widgets
Warning: QML import could not be resolved in any of the import paths: QtQuick.Extras.Private.CppUtils
Warning: QML import could not be resolved in any of the import paths: QtQuick.Extras.Private.CppUtils
  -- Skipping /home/qgc/Qt/5.15.2/android/plugins/position/libplugins_position_qtposition_serialnmea_arm64-v8a.so. It has unmet dependencies: lib/libQt5SerialPort_arm64-v8a.so.

FAILURE: Build failed with an exception.

* What went wrong:
Could not initialize class org.codehaus.groovy.runtime.InvokerHelper
> Exception java.lang.NoClassDefFoundError: Could not initialize class org.codehaus.groovy.reflection.ReflectionCache [in thread "Daemon worker"]

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 894ms
Building the android package failed!
  -- For more information, run this command with --verbose.
01:03:35: The process "/home/qgc/Qt/5.15.2/android/bin/androiddeployqt" exited with code 14.
Error while building/deploying project qgroundcontrol (kit: Android Qt 5.15.2 Clang Multi-Abi)
The kit Android Qt 5.15.2 Clang Multi-Abi has configuration issues which might be the root cause for this problem.
When executing step "Build Android APK"
01:03:36: Elapsed time: 00:22.

rebelboy, 2023-11-23:
[image]
[image]

rebelboy, 2023-11-23:
Ubuntu runs on vmware workstation 12

zdanek, 2023-11-23:
Hey.
Try to change jdk version. I had this ages ago.
Which one you have now?

rebelboy, 2023-11-24:
> […](#)

openjdk17
SDK Version: 11.0, NDK Version: 21.3.6528147

zdanek, 2023-11-26:
try to use JDK11
on Ubuntu I suggest using sdkman.io tool to easily install and switch.

provide feedback

rebelboy, 2023-11-27:
> try to use JDK11 on Ubuntu I suggest using sdkman.io tool to easily install and switch.
> 
> provide feedback

Thank You! I'll try it out

closed 2024-11-30 by HTRamsey