mavlink/qgroundcontrol #10880: 4.3.0 is available via docs.qgroundcontrol.com but not in GitHub releases

george-hawkins, 2023-11-29:
## Expected Behavior

I would expect, when clicking the `QGroundControl.dmg` link on the [Download and Install](https://docs.qgroundcontrol.com/master/en/getting_started/download_and_install.html) documentation page, that I'd get the same release as found at <https://github.com/mavlink/qgroundcontrol/releases/latest>

But the latest release today (Nov 29th, 2023) on GitHub is 4.2.9 while the link on the documentation site page is an Amazon CloudFront link - <https://d176tv9ibo4jno.cloudfront.net/latest/QGroundControl.dmg> - to a `.dmg` for version 4.3.0.

I'm a little uncomfortable using an unsigned executable that isn't clearly a product of the process that I can see on GitHub. Why this discrepancy between what's available via the documentation site and GitHub - surely any build is a product of the [GitHub actions](https://github.com/mavlink/qgroundcontrol/actions) so, the builds available on GitHub should lead those available via any other source?

## Log Files and Screenshots

Screenshot showing the link that I'm referring to:
[image]

DonLakeFlyer, 2023-11-29:
I don't know what the extra magic is to get a release marked on GitHub. @mrpollo Looks like you did last Stable. Could you do the same for the new 4.3?

mrpollo, 2023-11-30:
Just did, here's the link https://github.com/mavlink/qgroundcontrol/releases/tag/v4.3.0

george-hawkins, 2023-11-30:
Cool - very quickly resolved 👍

Might there be a way to automate this process, i.e. such that it's only possible for a new `.dmg` to end up being made available via the documentation site if it has first been marked as a release on GitHub?

DonLakeFlyer, 2023-11-30:
@mrpollo What's the magic to do this? I tried doing it myself by creating a release from the tag but I didn't get the artifacts?

DonLakeFlyer, 2023-11-30:
> We need to make a CI action that triggers either on a release or tag creation under a specific pattern

Already exists. That is how Stables/Daily builds get pushed up to S3.

> I usually run this manually after each tag, but it should be automated.

Run what? What are the steps to do it?

DonLakeFlyer, 2023-11-30:
Upload from tagged stable is here:
https://github.com/mavlink/qgroundcontrol/blob/master/.github/workflows/android_release.yml#L140

If you know how to do this in CI can you update?

mrpollo, 2023-11-30:
I don't have time to help right now, but you would need a step in which you check if it was triggered via tag creation, and make a new release if it's not already there, plus upload artifacts if not already present.